m5stack/M5GFX #10: m5core2 resets on load drawPng

Spacely70, 2021-07-19:
Hi there!

I have the problem that my m5core2 resets at displaying PNG files through drawPng:
`
#include <Arduino.h>

#include <M5GFX.h>
M5GFX display;

extern const uint8_t jpg[];

void setup()
{
// WiFi.begin("ssid","passwd");

  display.begin();

  if (display.width() < display.height())
  {
    display.setRotation(display.getRotation() ^ 1);
  }
}

void loop(void)
{

  display.drawPng(jpg, ~0u, 0, 0);
  delay(1000);

}

constexpr uint8_t jpg[] = {
0x89, 0x50, 0x4E, 0x47, 0x0D, 0x0A, 0x1A, 0x0A, 0x00, 0x00, 0x00, 0x0D, 0x49, 0x48, 0x44, 0x52, 
0x00, 0x00, 0x00, 0x40, 0x00, 0x00, 0x00, 0x40, 0x08, 0x06, 0x00, 0x00, 0x00, 0xAA, 0x69, 0x71, 
0xDE, 0x00, 0x00, 0x0A, 0x67, 0x49, 0x44, 0x41, 0x54, 0x78, 0x5E, 0xED, 0x9B, 0x5D, 0x6C, 0x55, 
0x55, 0x16, 0xC7, 0x4B, 0x41, 0x3E, 0x02, 0xFA, 0xE0, 0x84, 0x0A, 0x0C, 0xD0, 0x3A, 0x2D, 0x05, 
0x86, 0x0F, 0x95, 0x8F, 0x0E, 0x5A, 0x28, 0x30, 0xC8, 0x38, 0x0F, 0x1A, 0x79, 0xC5, 0x17, 0x32, 
0x13, 0x8D, 0x0F, 0x62, 0xA2, 0x0F, 0x26, 0x6A, 0xF4, 0x81, 0x79, 0xD0, 0x44, 0x3A, 0x4F, 0xC4, 
0x88, 0x99, 0x91, 0xD2, 0x96, 0xD6, 0x7E, 0xD1, 0x5B, 0x78, 0x41, 0x85, 0xC4, 0x21, 0xBC, 0x98, 
0xA8, 0x31, 0x0D, 0x46, 0x9D, 0x99, 0x60, 0xC4, 0x44, 0x8D, 0x98, 0x9E, 0x73, 0x2E, 0xA2, 0x78, 
0xC5, 0xEE, 0xF8, 0xDB, 0xFB, 0x9E, 0x7D, 0xEF, 0xBE, 0xE7, 0xE3, 0x9E, 0xCF, 0x3A, 0xC9, 0x44, 
0x93, 0x7F, 0x6A, 0xE2, 0x39, 0x7B, 0xEF, 0xDF, 0x5A, 0xEB, 0xEC, 0xBD, 0xD6, 0xDA, 0xD7, 0x86, 
0x86, 0x5F, 0xFF, 0xF9, 0xD5, 0x02, 0x33, 0x6A, 0x81, 0xBD, 0xA7, 0x8B, 0x37, 0xDF, 0x7B, 0xBA, 
0xB8, 0x17, 0x3D, 0x8B, 0x06, 0xF7, 0x9C, 0x2A, 0xBE, 0x83, 0x2E, 0xA3, 0x6B, 0x7F, 0x3C, 0x55, 
0xBC, 0x21, 0xB5, 0x9B, 0x7F, 0xDF, 0x3D, 0x51, 0xBC, 0xBC, 0x6B, 0xC2, 0x79, 0x07, 0x0D, 0xEE, 
0x9A, 0x28, 0x3E, 0xBB, 0x73, 0xC2, 0xD9, 0xDB, 0x35, 0x51, 0xBC, 0x79, 0x46, 0x17, 0x37, 0x53, 
0x83, 0x03, 0xBD, 0x04, 0x3D, 0x8E, 0xDE, 0x06, 0xBA, 0x84, 0x84, 0x14, 0xD0, 0x4A, 0x40, 0x57, 
0x04, 0xBC, 0x00, 0x5E, 0x00, 0xAD, 0xB4, 0xD3, 0x50, 0x57, 0xC1, 0x29, 0xA1, 0xB7, 0x77, 0x14, 
0x9C, 0xC7, 0xB7, 0x17, 0x8A, 0x4B, 0x66, 0x6A, 0xBD, 0xB9, 0x8C, 0xFB, 0xE7, 0x33, 0xD7, 0x1B, 
0x00, 0xEE, 0x44, 0xA3, 0xA8, 0x84, 0x14, 0xB4, 0x09, 0x6E, 0xC2, 0x7B, 0xC1, 0x4D, 0x78, 0xA0, 
0x85, 0x16, 0xF0, 0x42, 0x6A, 0x3B, 0xC6, 0xD8, 0x3E, 0xEE, 0x8C, 0x76, 0x8E, 0x3B, 0xF7, 0x6C, 
0x2B, 0x5C, 0xCF, 0x65, 0xCD, 0xB9, 0x0D, 0x02, 0x6C, 0x07, 0x3A, 0x27, 0xA1, 0x4D, 0xF0, 0x20, 
0xAF, 0x2B, 0xF0, 0xFA, 0x5E, 0x57, 0xF0, 0x06, 0xB8, 0x84, 0x17, 0xC0, 0x0B, 0xE0, 0x95, 0xEE, 
0x19, 0x77, 0xCE, 0xDE, 0x3D, 0xEE, 0x74, 0xE4, 0x06, 0x90, 0x76, 0x20, 0x60, 0x6F, 0xFD, 0xD3, 
0xE9, 0xE2, 0x6B, 0xFC, 0xFD, 0xC9, 0x0B, 0xAF, 0xC3, 0x3D, 0xA9, 0xD7, 0x35, 0xB8, 0xEB, 0x75, 
0x2F, 0xB8, 0x84, 0x57, 0xBA, 0xFB, 0xA4, 0xF3, 0x13, 0x7A, 0x6D, 0xDB, 0x49, 0xE7, 0xD6, 0xB4, 
0xEB, 0xCF, 0xF4, 0x1E, 0xE0, 0xF7, 0xA1, 0x2F, 0xB2, 0x7A, 0xBD, 0xAB, 0x50, 0x0C, 0x0A, 0xF7, 
0x1A, 0x70, 0xD7, 0xEB, 0x1A, 0x5C, 0xC2, 0x2B, 0x01, 0x2F, 0xF5, 0xC5, 0x1F, 0x4E, 0x3A, 0xF7, 
0x65, 0x82, 0x49, 0xF2, 0x32, 0xC0, 0x8D, 0xE8, 0x6F, 0x68, 0x3A, 0xCC, 0xEB, 0x09, 0x36, 0xB9, 
0x9A, 0x70, 0x8F, 0xE1, 0x75, 0x13, 0x5C, 0x00, 0x5E, 0xD6, 0x98, 0x33, 0xDD, 0x31, 0xE6, 0x1C, 
0xDA, 0x3A, 0x66, 0x37, 0x26, 0x61, 0x49, 0xFC, 0x2C, 0xC0, 0xF3, 0xD1, 0x48, 0x92, 0x70, 0x0F, 
0xDA, 0xE1, 0x43, 0x36, 0xB9, 0x48, 0xAF, 0xBB, 0x1E, 0x57, 0x9E, 0x77, 0xC1, 0x05, 0xE0, 0x15, 
0x6D, 0x1D, 0x73, 0x46, 0xD0, 0xFC, 0xC4, 0x60, 0x71, 0x5E, 0x00, 0x7A, 0x21, 0x3A, 0x1B, 0xB5, 
0xC9, 0x49, 0xEF, 0xFB, 0x37, 0x39, 0x47, 0x1E, 0x6F, 0x17, 0xD1, 0x2B, 0xC0, 0x3F, 0x8A, 0xF6, 
0xE0, 0xED, 0xF5, 0xA8, 0x65, 0x47, 0xA1, 0xD8, 0xC2, 0x26, 0xB7, 0x9E, 0x4D, 0x6E, 0x0F, 0xA1, 
0xFE, 0x28, 0x7A, 0x85, 0x6F, 0xFC, 0xA2, 0xFB, 0x9D, 0x7B, 0xC3, 0xBD, 0x02, 0x8E, 0xD7, 0x4D, 
0x70, 0x01, 0xB8, 0xD8, 0x32, 0xAA, 0x74, 0x76, 0xF3, 0xA8, 0xBD, 0x30, 0x0E, 0x53, 0xEC, 0x67, 
0x80, 0x9E, 0x87, 0xDE, 0x4A, 0x7E, 0xB4, 0x39, 0x9F, 0x00, 0xFD, 0x14, 0x6A, 0x06, 0x34, 0xF6, 
0x7C, 0xEC, 0xF0, 0x0D, 0x7C, 0xE3, 0xCD, 0xE8, 0x29, 0xBC, 0xFD, 0x89, 0xF6, 0xBC, 0xD7, 0xEB, 
0x12, 0x5A, 0xCB, 0x85, 0x17, 0x9B, 0x31, 0x02, 0x7A, 0x6B, 0xD3, 0xA8, 0x33, 0x2F, 0xF6, 0x84, 
[... 114 lines cut ...]
0xC8, 0xAF, 0x02, 0xF8, 0x86, 0x55, 0x83, 0xD6, 0x5F, 0x57, 0x0D, 0x58, 0xDF, 0x9B, 0xE0, 0x61, 
0xFD, 0xF7, 0x88, 0xD6, 0x94, 0xB7, 0x56, 0x0F, 0x85, 0xAF, 0xEF, 0x75, 0xEB, 0x7B, 0xE0, 0xFF, 
0xD2, 0xD4, 0x63, 0x47, 0xAE, 0x3F, 0xB7, 0x07, 0x00, 0xDE, 0xD8, 0x36, 0x60, 0x4D, 0x06, 0x81, 
0x07, 0x75, 0x62, 0x43, 0x5A, 0x53, 0xA9, 0xBC, 0xEE, 0x6E, 0x72, 0xEA, 0x7B, 0x6F, 0xEA, 0xB1, 
0x26, 0x81, 0xDF, 0x98, 0x1B, 0x58, 0x92, 0x81, 0xD8, 0xE0, 0xE6, 0x92, 0xBF, 0x3F, 0xC7, 0xDF, 
0x6B, 0xE6, 0xAD, 0x4B, 0x98, 0xD7, 0x03, 0x5A, 0x53, 0x3E, 0x8F, 0x07, 0x6D, 0x72, 0xD5, 0xA3, 
0x4D, 0x6F, 0x72, 0xEA, 0x5B, 0xBF, 0x06, 0xF8, 0x73, 0x28, 0x97, 0x2B, 0xBB, 0x24, 0xDC, 0xBE, 
0x67, 0x81, 0x5F, 0x89, 0xFE, 0x09, 0x78, 0xA9, 0x4E, 0x27, 0x36, 0x76, 0xB8, 0xBB, 0xC5, 0x8B, 
0x99, 0xD0, 0xE8, 0xA3, 0x4D, 0xEE, 0xF0, 0x25, 0xA0, 0xFF, 0x81, 0xE7, 0x57, 0x66, 0x5A, 0xF4, 
0x4C, 0xBC, 0x7C, 0x7B, 0xBF, 0xD5, 0x4C, 0xA8, 0x1F, 0x46, 0x57, 0x8C, 0x16, 0x74, 0x0D, 0x78, 
0x40, 0x93, 0x22, 0x2C, 0x7F, 0xF7, 0x1C, 0x6D, 0xD6, 0x15, 0xC0, 0x0F, 0x03, 0xDE, 0x3C, 0x13, 
0x6B, 0xCF, 0x75, 0xCC, 0xE6, 0x3E, 0x6B, 0x1E, 0x06, 0x78, 0x80, 0x90, 0x3F, 0xC6, 0xF1, 0xF6, 
0xA5, 0xB7, 0x4E, 0x0F, 0x4A, 0x68, 0x42, 0x36, 0xB9, 0x2F, 0xF9, 0xCE, 0x8F, 0xDD, 0xD6, 0x63, 
0x3D, 0x00, 0x78, 0x3E, 0xFF, 0x07, 0x48, 0xAE, 0xA4, 0x31, 0x06, 0xA3, 0x03, 0xDB, 0x80, 0x01, 
0xDA, 0x81, 0xDE, 0x8F, 0x5E, 0xE0, 0x1B, 0x1F, 0x42, 0xE7, 0x09, 0xF3, 0x8B, 0xE8, 0xBF, 0x52, 
0xC0, 0x5F, 0xE4, 0x3B, 0x3F, 0xCF, 0x91, 0x36, 0x84, 0x5E, 0x20, 0x8D, 0xDD, 0xCF, 0x37, 0xDE, 
0xBE, 0xE4, 0x97, 0xDC, 0xD5, 0x63, 0xB0, 0xFC, 0x5F, 0x3E, 0xF2, 0x33, 0x10, 0xEB, 0x46, 0xD1, 
0xED, 0xC8, 0x90, 0xFD, 0x00, 0x00, 0x00, 0x24, 0x74, 0x45, 0x58, 0x74, 0x43, 0x6F, 0x70, 0x79, 
0x72, 0x69, 0x67, 0x68, 0x74, 0x00, 0x43, 0x6F, 0x70, 0x79, 0x72, 0x69, 0x67, 0x68, 0x74, 0x20, 
0x32, 0x30, 0x31, 0x35, 0x20, 0x49, 0x4E, 0x43, 0x4F, 0x52, 0x53, 0x20, 0x47, 0x6D, 0x62, 0x48, 
0x7A, 0x86, 0xDD, 0x36, 0x00, 0x00, 0x00, 0x00, 0x49, 0x45, 0x4E, 0x44, 0xAE, 0x42, 0x60, 0x82
};`

i'm using the actual Arduino IDE 1.8.15
Libraries are up to date (updated today).

lovyan03, 2021-07-19:
@Spacely70 
Thank you for reporting this issue !
I have made a fix in the develop branch.
If you want to try it right away, you can get it from the develop branch.

Spacely70, 2021-07-19:
Hi lovyan03,

yes...working now!

Thanks!

closed 2021-07-24 by lovyan03