m5stack/M5GFX #25: LCD Unit 1.14 + M5StickC PLUS

shamilton55, 2021-12-03:
Hi, I would like to know if there is a way to duplicate the same image that shows on the M5StickC PLUS screen to LCD Screen Unit 1.14

shamilton55, 2021-12-03:
The two images must be appearing at the same time

lovyan03, 2021-12-03:
@shamilton55 
Since they are independent screens, you will need to draw on both screens by yourself.

I think it is easy to draw off-screen using M5Canvas and pushSprite to both.
However, keep in mind that this method uses a lot of memory and the drawing speed on the UnitLCD is not very fast.

The recommendation is to prepare a function that receives a pointer to the base class LovyanGFX and draw on both.

```
#include <M5GFX.h>
#include <M5UnitLCD.h>

M5GFX lcd1;
M5UnitLCD lcd2(32, 33); // SDA=G32, SCL=G33

void draw(m5gfx::LovyanGFX* gfx)
{
  gfx->fillRect(50, 50, 50, 50, TFT_BLUE);
  gfx->fillCircle(40, 40, 20, TFT_GREEN);
  gfx->print("Hello World");
}

void setup()
{
  lcd1.init();
  lcd2.init();

  draw(&lcd1); // draw to M5StickCPlus
  draw(&lcd2); // draw to UnitLCD
}

void loop() {}
```

lovyan03, 2021-12-03:
@shamilton55 
If speed is not so important, you can also read the pixels from the LCD of the M5StickCPlus and transfer them to the UnitLCD.

```
#include <M5GFX.h>
#include <M5UnitLCD.h>

M5GFX lcd1;
M5UnitLCD lcd2(32, 33); // SDA=G32, SCL=G33

void copy(m5gfx::LovyanGFX* dst, m5gfx::LovyanGFX* src)
{
  auto buf = (m5gfx::swap565_t*)alloca(src->width() * sizeof(m5gfx::swap565_t));
  src->startWrite();
  dst->startWrite();
  for (size_t y = 0; y < src->height(); ++y)
  {
    src->readRect(0, y, src->width(), 1, buf);
    dst->pushRect(0, y, src->width(), 1, buf);
  }
  src->endWrite();
  dst->endWrite();
}

void setup(void)
{
  lcd1.init();
  lcd2.init();
}

void loop(void)
{
  lcd1.fillRect(rand()&127, rand()&127, 32, 32, rand());
  copy(&lcd2, &lcd1);
  delay(1);
}
```

shamilton55, 2021-12-03:
@lovyan03 I created a tally light with M5Stack PLUS for the cameras to inform who is in the air, I want to put another screen attracts for the presenter to see which camera is in the air too

**Project image**
[https://scontent.fcgb1-1.fna.fbcdn.net/v/t39.30808-6/262012745_4473154242771848_5767250440311476716_n.jpg?_nc_cat=104&ccb=1-5&_nc_sid=dbeb18&_nc_eui2=AeESgKcggNJjeIP2wINF6A5RvcCatngpl9O9wJq2eCmX023cHhqKvinzbU81VnDcMh5Ki5WRHYEcDB2f5-Uy90Oq&_nc_ohc=mJSwh0jng3EAX-LAlp0&_nc_ht=scontent.fcgb1-1.fna&oh=71c5a45d6f99b4a318372c15f364792b&oe=61AE08D8](url)

shamilton55, 2021-12-03:
#include <M5StickCPlus.h>
#include <WiFi.h>

#include <SkaarhojPgmspace.h>
#include <ATEMbase.h>
#include <ATEMstd.h>

// Define the IP address of your ATEM switcher
IPAddress switcherIp(192, 168, 3, 107);

// Put your wifi SSID and password here
const char* ssid = "useratem";
const char* password = "a01atem";

// Set this to 1 if you want the orientation to update automatically
#define AUTOUPDATE_ORIENTATION 1

// You can customize the red/green/grey if you want
// http://www.barth-dev.de/online/rgb565-color-picker/
#define GRAY  0x0841 //   8   8  8
//#define GREEN 0x0400 //   0 128  0
#define RED   0xF800 // 255   0  0

/////////////////////////////////////////////////////////////
// You probably don't need to change things below this line
#define LED_PIN 10

ATEMstd AtemSwitcher;

int orientation = 0;
int orientationPrevious = 0;
int orientationMillisPrevious = millis();
int buttonBMillis = 0;

int cameraNumber = 1;
int cameraTotal = 8;

int previewTallyPrevious = 1;
int programTallyPrevious = 1;
int cameraNumberPrevious = cameraNumber;

void setup() {
  Serial.begin(115200);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");

  // 初期化
  M5.begin();
  M5.imu.Init();
  M5.Lcd.setRotation(orientation);

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  AtemSwitcher.begin(switcherIp);
  AtemSwitcher.serialOutput(0x80);
  AtemSwitcher.connect();

  // GPIO初期化
  pinMode(26, INPUT); // PIN  (INPUT, OUTPUT, ANALOG)無線利用時にはANALOG利用不可, DAC出力可
  pinMode(36, INPUT); // PIN  (INPUT,       , ANALOG)入力専用、INPUT_PULLUP等も不可
  pinMode( 0, INPUT); // PIN  (INPUT, OUTPUT,       )外部回路でプルアップ済み
  pinMode(32, INPUT); // GROVE(INPUT, OUTPUT, ANALOG)
  pinMode(33, INPUT); // GROVE(INPUT, OUTPUT, ANALOG)
}

void setOrientation() {
  float accX = 0, accY = 0, accZ = 0;
  M5.imu.getAccelData(&accX, &accY, &accZ);
  //Serial.printf("%.2f   %.2f   %.2f \n",accX * 1000, accY * 1000, accZ * 1000);

  if (accZ < .9) {
    if (accX > .6) {
      orientation = 1;
    } else if (accX < .4 && accX > -.5) {
      if (accY > 0) {
        orientation = 0;
      } else {
        orientation = 2;
      }
    } else {
      orientation = 3;
    }
  }

  if (orientation != orientationPrevious) {
    Serial.printf("Orientation changed to %d\n", orientation);
    M5.Lcd.setRotation(orientation);
  }
}

void loop() {
  // ボタンの状態更新
  M5.update();

  if (AUTOUPDATE_ORIENTATION) {
    if (orientationMillisPrevious + 500 < millis()) {
      setOrientation();
      orientationMillisPrevious = millis();
    }
  }

  if (M5.BtnA.wasPressed()) {
    AtemSwitcher.changeProgramInput(cameraNumber);
  }
  if (M5.BtnB.wasPressed()) {
    setOrientation();
    buttonBMillis = millis();
  }

  if (M5.BtnB.isPressed() && buttonBMillis != 0 && buttonBMillis < millis() - 500) {
    Serial.println("Changing camera number");
    cameraNumber = (cameraNumber % cameraTotal) + 1;
    Serial.printf("New camera number: %d\n", cameraNumber);

    buttonBMillis = 0;
  }

  // Check for packets, respond to them etc. Keeping the connection alive!
  AtemSwitcher.runLoop();

  int programTally = AtemSwitcher.getProgramTally(cameraNumber);
  int previewTally = AtemSwitcher.getPreviewTally(cameraNumber);

  if ((orientation != orientationPrevious) || (cameraNumber != cameraNumberPrevious) || (programTallyPrevious != programTally) || (previewTallyPrevious != previewTally)) { // changed?
    if (programTally && !previewTally) { // only program
      drawLabel(RED, BLACK, LOW);
    } else if (programTally && previewTally) { // program AND preview
      drawLabel(RED, GREEN, LOW);
    } else if (previewTally && !programTally) { // only preview
      drawLabel(GREEN, BLACK, HIGH);
    } else if (!previewTally || !programTally) { // neither
      drawLabel(BLACK, GRAY, HIGH);
    }
  }

  programTallyPrevious = programTally;
  previewTallyPrevious = previewTally;
  cameraNumberPrevious = cameraNumber;
  orientationPrevious  = orientation;
}

void drawLabel(unsigned long int screenColor, unsigned long int labelColor, bool ledValue) {
  digitalWrite(LED_PIN, ledValue);
  M5.Lcd.fillScreen(screenColor);
  M5.Lcd.setTextColor(labelColor, screenColor);
  drawStringInCenter(String(cameraNumber), 8);
}

void drawStringInCenter(String input, int font) {
  int datumPrevious = M5.Lcd.getTextDatum();
  M5.Lcd.setTextDatum(MC_DATUM);
  M5.Lcd.drawString(input, M5.Lcd.width() / 2, M5.Lcd.height() / 2, font);
  M5.Lcd.setTextSize(2);
  M5.Lcd.setTextDatum(datumPrevious);
  
}

shamilton55, 2021-12-03:
> @ lovyan03 Criei uma luz de contagem com M5Stack PLUS para as câmeras informarem quem está no ar, quero colocar outra tela atrai para o apresentador ver qual câmera está no ar também
> 
> **Imagem do projeto ** https://ibb.co/c8z4J9d

closed 2021-12-03 by shamilton55