m5stack/M5GFX #28: Request: Writing QOI images

prjh, 2022-01-29:
I know that QOI is not already a standard, and an IANA type for that new image format is on the way. But it has a great potential especially for small devices. 
So, as I have seen, that the LovyanGFX already supports createQoi() in the Qoi branch, I want kindly ask, if it is possible to merge that also to the M5GFX repo. For me M5Unified + M5GFX has alot of pro's on the M5Core2 (and other) M5 plattforms. 
At this point, a big thanks to the high-quality contribution of M5GFX and M5Unified.

lovyan03, 2022-01-30:
@prjh 
Yes, I do have a plan.
I'll be maintaining the M5GFX to keep up with the LovyanGFX release.
If LovyanGFX adopts it, it will be adopted by M5GFX within a few days.

Please wait a bit, as we have a lot of work piling up that has a higher priority than the QOI save feature…

prjh, 2022-01-30:
@lovyan03 
I understand that, no problem. I have already noted with pleasure that there is real life in the new M5GFX project. Good things take time. By the way, creating and drawing from memory would also be a fine feature, but no hurry ;-)
... and big THX