m5stack/M5GFX #30: Where to find the original example, that is running on the ATOM Display, when shipped?

Fischkopppp, 2022-03-25:
Hi, thank you for the great work. I am looking for the original example, that is running on the ATOM Display after first startup. I am interested in the PDQ graphicstest part, as it supports for example roundRects. I have a few existing codes, that I would like to run  on the ATOM Display, however they are built on the TFT_eSPI library. For my understanding I am able to use TFT_eSPI functions by adding: 

#include <LGFX_TFT_eSPI.hpp>
//#include <TFT_eSPI.h>

However I was not successful to get this running, or even compiling. This is why I thought I could understand how to do it, with the original example, that is running on the Atom Display, as I can see that is basically running the TFT_graphicstest_PDQ.ino example.

Thank you for any help or suggestions.

lovyan03, 2022-03-26:
@Fischkopppp 
Thank you for your report, I have added the sample in the develop branch.
It will take a few days to be reflected in the master branch and release, but if you want to try it immediately, please get it from the develop branch.

M5GFX encapsulates the LovyanGFX core, but LGFX_TFT_eSPI.hpp is not available.
To use AtomDisplay, do the following

```
#include <M5AtomDisplay.h>
M5AtomDisplay display;

void setup(void)
{
  display.init();

  /// Use a function to draw something.
  display.fillRect(10, 20, 30, 40, TFT_BLUE);
}
```

Fischkopppp, 2022-03-27:
Thank you for the reply! Are you planning to add LGFX_TFT_eSPI.hpp in the future?
I will try the example from the develop branch.

lovyan03, 2022-03-27:
@Fischkopppp 
sorry. There are no plans to add the same functionality as LGFX_TFT_eSPI to M5GFX.
If you look at the contents of the file, you will see that I have only simple constant definitions.
I recommend that you copy and use only those parts of the file that you need for your project, if necessary.

Fischkopppp, 2022-03-27:
@lovyan03  
thank you for the quick reply. I added fillSmoothRoundRect and fillSmoothCircle to LGFXBase.cpp and LGFXBase.hpp, this way the project is compiling now. I just copied the code from fillRoundRect / fillCircle.

> I recommend that you copy and use only those parts of the file that you need for your project, if necessary.

Do you mean to simply use the LGFX_TFT_eSPI.hpp part in my projects? I will try that, too.

lovyan03, 2022-03-27:
@Fischkopppp 
wow... I was not aware that fillSmoothRoundRect and fillSmoothCircle had been added.
I will consider adopting them. thank you !

closed 2025-05-16 by lovyan03