m5stack/M5GFX #36: M5unitLCD for ESP32 I2C2 issue

NathanKuo0821, 2022-04-26:
Hi All,

I have defined two I2Cs in ESP32 to control M5UnitLCD.
Now there is a problem, when i use I2C1 to control LCD everything works fine, but when i switch to I2C2, I2C error occurs, and i use gpio 15/16 or 32/33 to control I2C1 without problem, but I2C2 LCD has problems , can anyone help?
Thansk 

**ERROR LOG:**
[     1][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 MhPB
[    20][E][common.cpp:939] writeBytes(): i2c write error : ack wait
[   219][E][common.cpp:939] writeBytes(): i2c write error : ack wait
[   219][D][main.cpp:498] setup(): width=0 , height=0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception wasandled.

Core  1 register dump:
PC      : 0x400de05f  PS      : 0x00060930  A0      : 0x800de646  A1      : fb26f0
A2      : 0x00002710  A3      : 0x3ffbe02c  A4      : 0x00000032  A5      : 027000
A6      : 0x00000020  A7      : 0x00004e20  A8      : 0x800de055  A9      : fb26d0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x60027000  A13     : 0007bf
A14

**CODE:**
//I2C BUS
#define I2C1_SDA 15
#define I2C1_SDL 16
#define I2C2_SDA 32
#define I2C2_SDL 33

/* I2C Bus */
TwoWire I2C1 = TwoWire(0); //I2C1 bus
TwoWire I2C2 = TwoWire(1); //I2C2 bus

// LCD
#include <M5UnitLCD.h>
M5UnitLCD display (I2C2_SDA,I2C2_SDL,400000); 
void setup()
{
  Serial.begin(115200);
 I2C2.begin(I2C2_SDA,I2C2_SDL);

   display.begin();

  if (display.isEPD())
  {
    display.setEpdMode(epd_mode_t::epd_fastest);
    display.invertDisplay(true);
    display.clear(TFT_WHITE);
  }
  if (display.width() < display.height())
  {
    display.setRotation(display.getRotation() ^ 1);
  }

  // width=240 , height=135
  log_d("width=%d , height=%d",display.width(),display.height());

  display.setColorDepth(1); // mono color
  display.setFont(&fonts::Font4);
  display.setTextWrap(false);
  display.setTextSize(3);
  display.setTextColor(TFT_BLACK);

  display.fillScreen(WHITE);

  display.drawString("Ready" , 0, (display.height()/3));
}

Nathan Kuo

lovyan03, 2022-04-26:
@NathanKuo0821 
The constructor of M5UnitLCD and M5UnitOLED can set the I2C port number in the fourth argument.
Try the following

```
  M5UnitLCD display (I2C2_SDA,I2C2_SDL,400000, I2C_NUM_1);
```

NathanKuo0821, 2022-04-27:
> @NathanKuo0821 The constructor of M5UnitLCD and M5UnitOLED can set the I2C port number in the fourth argument. Try the following
> 
> ```
>   M5UnitLCD display (I2C2_SDA,I2C2_SDL,400000, I2C_NUM_1);
> ```

Dear Sir,
Thanks for your reply .
When I set the I2C_NUM_1 in M5UnitLCD as below , ESP32 keeps crashing and restarting.
Is this a bug for M5UnitLCD using ESP32 I2C2?

`M5UnitLCD display (I2C2_SDA,I2C2_SDL,400000, I2C_NUM_1); `

Error Log:
[E][common.cpp:939] writeBytes(): i2c write error : ack wait
[E][common.cpp:939] writeBytes(): i2c write error : ack wait
[D][DetectionManage.cpp:148] displayinit(): width=0 , height=0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e1ba6  PS      : 0x00060f30  A0      : 0x800e207a  A1      : 0x3ffb1c00
A2      : 0x00000020  A3      : 0x00000000  A4      : 0x00000032  A5      : 0x3ffbe218
A6      : 0x60027000  A7      : 0x00004e20  A8      : 0x800e1b9e  A9      : 0x3ffb1be0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x000007bf
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x400e1ba6:0x3ffb1c00 0x400e2077:0x3ffb1c40 0x400d9a15:0x3ffb1c60 0x400d9a75:0x3ffb1c80 0x400ddd3f:0x3ffb1ca0 0x400ddda9:0x3ffb1cc0 0x400d2580:0x3ffb1ce0 0x400d44a3:0x3ffb1d00 0x400e2cae:0x3ffb1fb0 0x4008a33a:0x3ffb1fd0

Nathan Kuo

lovyan03, 2022-04-27:
@NathanKuo0821 
What version of ArduinoESP32 do you have?
Is the M5GFX up-to-date?

NathanKuo0821, 2022-04-27:
> @NathanKuo0821 What version of ArduinoESP32 do you have? Is the M5GFX up-to-date?

The version as below , Is this ok?
framework-arduinoespressif32 3.20000.220418 (2.0.0)
m5stack/M5GFX@^0.0.20

**Error Log:**
[    18][E][common.cpp:1051] writeBytes(): i2c write error : ack wait
[    19][D][DetectionManage.cpp:148] displayinit(): width=0 , height=0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400eb5f7  PS      : 0x00060930  A0      : 0x800ebbf6  A1      : 0x3ffb2460
A2      : 0x00002710  A3      : 0x3ffbe158  A4      : 0x00000032  A5      : 0x60027000
A6      : 0x00000020  A7      : 0x00004e20  A8      : 0x800eb5ed  A9      : 0x3ffb2440
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffb99c4  A13     : 0x3ffc38a8
A14     : 0x3ffb9904  A15     : 0x3ffb991c  SAR     : 0x00000010  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x40089234  LEND    : 0x4008923f  LCOUNT  : 0x00000000

Backtrace:0x400eb5f4:0x3ffb24600x400ebbf3:0x3ffb24a0 0x400de61d:0x3ffb24d0 0x400de879:0x3ffb24f0 0x400de8e5:0x3ffb2510 0x400d4ec0:0x3ffb2530 0x400d69df:0x3ffb2570 0x400ec5e7:0x3ffb2820

Nathan Kuo

NathanKuo0821, 2022-04-27:
If I switch to "I2C1" pin and "I2C_NUM_0" , everything works fine.

`M5UnitLCD display (I2C1_SDA,I2C1_SDL,400000, I2C_NUM_0);`

or change I2C1 pin to gpio 32/33 and "I2C_NUM_0" , it also still works.

```
/*
#define I2C1_SDA 15
#define I2C1_SDL 16
#define I2C2_SDA 32
#define I2C2_SDL 33
*/
#define I2C1_SDA 32
#define I2C1_SDL 33
#define I2C2_SDA 15
#define I2C2_SDL 16

M5UnitLCD display (I2C1_SDA,I2C1_SDL,400000, I2C_NUM_0);
```
I have no idea for this , please give me some more advice.
Thanks again.

lovyan03, 2022-04-27:
@NathanKuo0821 
The TwoWire of the ArduinoESP32 has a prepared instance.
Perhaps your creating a separate instance would cause inconvenience.
Try using `Wire` instead of `I2C1` and `Wire1` instead of `I2C2`.

NathanKuo0821, 2022-04-27:
> @NathanKuo0821 The TwoWire of the ArduinoESP32 has a prepared instance. Perhaps your creating a separate instance would cause inconvenience. Try using `Wire` instead of `I2C1` and `Wire1` instead of `I2C2`.

I also tried the example "ESP32 Using Two I2C Bus Interfaces" in this guide "https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/" using I2C1 and I2C2 to control other sensors and it works fine.
So I don't know what else I can do to verify I2C1 and I2C2 in our device.

Nathan Kuo

NathanKuo0821, 2022-04-27:
> @NathanKuo0821 The TwoWire of the ArduinoESP32 has a prepared instance. Perhaps your creating a separate instance would cause inconvenience. Try using `Wire` instead of `I2C1` and `Wire1` instead of `I2C2`.
Dear Sir,

You are right, I2C2 is working.
We can close this thread.
Thanks again for your help.

Nathan kuo

closed 2022-04-27 by NathanKuo0821