m5stack/M5GFX #47: Using AtomDisplay with Makerfabs 1920x480 IPS display with MIPI-HDMI Driver

Daqaco, 2023-03-24:
Recently got a number of these displays [link](https://www.makerfabs.com/8-8-inch-hdmi-ips-display.html) but cannot get them to work with the M5Stack AtomDisplay though I do see some artefacts displayed on the screen when I run some test code with Arduino IDE.  I know that they are 60Hz with a pixel clock of 66280000.  I can drive them with other HDMI sources such as my development Mac, embedded computers etc but not with the AtomDisplay.  I'm not a display kinda guy, I write industrial control and state machine code :-). Just wondering if these will work with the AtomDisplay and if so, any pointers to getting them to work?  Really appreciate any help + apologies if this is not the right place to post this question.

I'm using the AtomDisplay with PSRAM though I know this makes no difference so the controller is an ESP32 PICO-V3

Mark

lovyan03, 2023-03-24:
Hello, @Daqaco .
I don't have that display, but presumably the correct resolution is 480 (width) x 1920 (height).
Try the following...

```
#include <M5AtomDisplay.h>

M5AtomDisplay display ( 480, 1920, 60, 480, 1920, 1, 1, 66280000 );

void setup(void)
{
  display.init();

// Set the display orientation
// 0 = 270 deg / 1 = normal / 2 = 90 deg / 3 = 180 deg / 4~7 = upside down
  display.setRotation(1);

  display.setTextSize(5);
  display.printf("Hello, ATOM Display !");
}
```

Daqaco, 2023-03-28:
Have been in dialog with the supplier of the display and got some specs from them on the display and the HDMI-MIPI interface but no joy in getting an output.  I had read that the limit is 55,296,000 and 1920 x 480 x 60 is 55,296,000.  I'm going to source a couple of other interface boards to try out but for now I'll continue the development I need to do using a standard HDMI monitor which work really well with the M5Stack AtomDisplay.  Thanks for the help though!

closed 2023-03-28 by Daqaco