m5stack/M5GFX #51: M5AtomDisplay working with pre-installed software but not with the software from Github

Onzeisnottwice, 2023-05-17:
So, this is the situation.
I have two M5AtomDisplay Units, say A and B and I have a wave share 5 inch HDMI display (https://www.waveshare.com/wiki/5inch_HDMI_LCD#Screen_orientation_settings).

Unit A is out of the box plugged to my Waveshare screen. It rins smoothly the graphics. To clarify, I have not uploaded any software to the M5Atom.

Unit B I uploaded the software (Atomdisplay_factory) to the M5Atom. I use Arduino IDE, I dont see any related libs to be updated. BUT, nog graphics shown on the screen. I tried different setting for M5AtomDisplay display.

Anyone any clue how to get this Waveshare working with M5AtomDisplay. Oh, btw, unit B works perfectly with the Waveshare 7 inch display. Help is appreciated.

lovyan03, 2023-05-18:
Hello, @Onzeisnottwice 
ModuleDisplay and AtomDisplay is not officially HDMI compliant, so special resolution settings may not be possible.

The timing parameters are determined by program calculations, but there is no way to calculate the correct parameters, They are determined by my intuition.
The formulas have been changed several times, so it is possible that they are no longer displayed properly in the update.

BTW, Related Issue. ( #48 )

He says he can't get Waveshare 7" to work, can you publish what program you have written?

Onzeisnottwice, 2023-05-19:
Hello lovyan03, Thanks for having attention to this issue. Before come into your remark, please let me for clarity make 2 definitions.
AtomDisplay A. This is the M5AtomDisplay with the factory installed software. So I have not uploaded any software at all.
AtomDisplay B. This is the M5AtomDisplay with demo software known as AtomDisplay_factory installed by me.
I am using Arduino IDE 2.1.0 and M5GFX 0.1.6 installed. This M5GFX 0.1.6 also provides me the earlier mentioned AtomDisplay_factory demo software which I have installed on AtomDisplay B.
I also have two different Waveshare screens: https://www.waveshare.com/wiki/7inch_HDMI_LCD_(C) (7 inch) and https://www.waveshare.com/wiki/5inch_HDMI_LCD#Screen_orientation_settings (5 inch). And now this is the issue. AtomDisplay A works on both screens. AtomDisplay B only works on the 7 inch screen, whatever setting I use in the display call. 
I have attached two photos showing you that both screen work on AtomDisplay A. Referring to your question from issue #48 the M5GFX factory demo software installed by me on AtomDisplay B works neatly on the 7 inch screen. So modification done by me in that software. Btw the size is in centimetres (17 cm is 7 inch). In your explanation you mention that certain calculations have been changed during the upgrading of the M5GFX library. So one explanation of my issue is, is that on AtomDispplay A calculation are made that work for both displays while within the current version of the M5GFX version the calculations have been changed which a a result leads to the 5inch not working anymore. I have also included the startup messages of AtomDisplay A. Maybe that might help in resolving this issue. For example which version of M5GFX is shipped with AtomDisplay A. A long story but I hope it might help you to help me. 

Startup messages from AtomDisplay A:
M5AtomDisplay

ets Jun  8 2016 00:22:57
08:56:12.201 -> 
08:56:12.201 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
08:56:12.201 -> configsip: 188777542, SPIWP:0xee
08:56:12.201 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
08:56:12.201 -> mode:DIO, clock div:1
08:56:12.201 -> load:0x3fff0030,len:7388
08:56:12.201 -> ho 0 tail 12 room 4
08:56:12.201 -> load:0x40078000,len:15464
08:56:12.201 -> ho 0 tail 12 room 4
08:56:12.201 -> load:0x40080400,len:4452
08:56:12.201 -> entry 0x40080694
08:56:12.201 -> [0;32mI (31) boot: ESP-IDF v4.3 2nd stage bootloader[0m
08:56:12.233 -> [0;32mI (31) boot: compile time 20:58:19[0m
08:56:12.233 -> [0;32mI (31) boot: chip revision: 1[0m
08:56:12.233 -> [0;32mI (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
08:56:12.233 -> [0;32mI (41) qio_mode: Enabling default flash chip QIO[0m
08:56:12.233 -> [0;32mI (46) boot.esp32: SPI Speed      : 80MHz[0m
08:56:12.233 -> [0;32mI (51) boot.esp32: SPI Mode       : QIO[0m
08:56:12.282 -> [0;32mI (55) boot.esp32: SPI Flash Size : 4MB[0m
08:56:12.282 -> [0;32mI (60) boot: Enabling RNG early entropy source...[0m
08:56:12.282 -> [0;32mI (65) boot: Partition Table:[0m
08:56:12.282 -> [0;32mI (69) boot: ## Label            Usage          Type ST Offset   Length[0m
08:56:12.282 -> [0;32mI (76) boot:  0 nvs              WiFi data        01 02 00009000 00005000[0m
08:56:12.282 -> [0;32mI (84) boot:  1 otadata          OTA data         01 00 0000e000 00002000[0m
08:56:12.297 -> [0;32mI (91) boot:  2 app0             OTA app          00 10 00010000 00300000[0m
08:56:12.297 -> [0;32mI (99) boot:  3 spiffs           Unknown data     01 82 00310000 000f0000[0m
08:56:12.297 -> [0;32mI (106) boot: End of partition table[0m
08:56:12.297 -> [0;32mI (110) boot_comm: chip revision: 1, min. application chip revision: 0[0m
08:56:12.297 -> [0;32mI (110) boot_comm: chip revision: 1, min. application chip revision: 0[0m
08:56:12.297 -> [0;32mI (118) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=700fch (459004) map[0m
08:56:12.462 -> [0;32mI (264) esp_image: segment 1: paddr=00080124 vaddr=3ffbdb60 size=03c68h ( 15464) load[0m
08:56:12.462 -> [0;32mI (269) esp_image: segment 2: paddr=00083d94 vaddr=40080000 size=0c27ch ( 49788) load[0m
08:56:12.462 -> [0;32mI (287) esp_image: segment 3: paddr=00090018 vaddr=400d0018 size=7f0f0h (520432) map[0m
08:56:12.624 -> [0;32mI (443) esp_image: segment 4: paddr=0010f110 vaddr=4008c27c size=03c2ch ( 15404) load[0m
08:56:12.624 -> [0;32mI (458) boot: Loaded app from partition at offset 0x10000[0m
08:56:12.657 -> [0;32mI (458) boot: Disabling RNG early entropy source...[0m
08:56:12.690 -> E (25) M5HDMI: i2c port:1 sda:25 scl:21
08:56:12.690 -> E (27) M5HDMI: Chip ID: 17 02 e2
08:56:12.690 -> 
08:56:12.690 -> 
08:56:12.690 -> E (27) M5HDMI: Resetting HDMI transmitter...
08:56:12.690 -> E (29) M5HDMI: Erase FPGA SRAM...
08:56:12.690 -> E (37) M5HDMI: Starting Writing to SRAM...
08:56:13.186 -> E (514) M5HDMI: SRAM Prog Finish...
08:56:13.186 -> E (514) M5HDMI: ---------------------------
08:56:13.186 -> E (514) M5HDMI: Waiting for DONE...
08:56:13.186 -> E (515) M5HDMI: Done!
08:56:13.186 -> E (516) M5HDMI: ---------------------------
08:56:13.186 -> E (519) M5HDMI: Total Time: 482 ms
08:56:13.186 -> E (524) M5HDMI: Initialize HDMI transmitter...
08:56:13.186 -> E (529) M5HDMI: V: polarity=0, width=5, back=20, front=5, total=750, active=720
08:56:13.218 -> E (534) M5HDMI: H: polarity=0, width=40, back=220, front=110, total=1650, active=1280
[... cut ...]
08:56:13.218 -> E (555) M5HDMI: 	#0: locked=1,value=01,done=1
08:56:13.218 -> 
08:56:13.218 -> E (555) M5HDMI: 	#0: locked
08:56:13.218 -> 
08:56:13.218 -> E (558) M5HDMI: done.
08:56:13.218 -> E (558) M5HDMI: Waiting the FPGA gets idle...
08:56:13.218 -> E (558) M5HDMI: Reading FPGA ID... 
08:56:13.251 -> E (561) M5HDMI: FPGA ID:48 44 00 02
08:56:13.251 -> 
08:56:59.297 -> [I][AtomDisplayDEMO.cpp:954] loop(): HaD pushColor            187613
08:57:00.088 -> [I][AtomDisplayDEMO.cpp:959] loop(): Screen fill              53717

[image]
[image]

Onzeisnottwice, 2023-05-19:
So I have done another experiment with AtomDisplay B. I have tried all the old versions of the M5GFX lib out with the demo software for the AtomDisplay. None of them shows anything on the 5inch Waveshare screen. I hope thes information might be useful to allocate the source problem. For example, could it be that the root cause is not something that has changed in the M5GFX lib but in some kind of other library?

lovyan03, 2023-05-19:
@Onzeisnottwice 
The shipping demonstration was built with a version of the M5GFX that was still under development prior to its official release.

To explain the situation, I needed firmware for a demonstration to show at a trade show before the product was released, so I built it using the M5GFX at a time when it had not been officially released.
That firmware is now the firmware for shipping.

BTW, the factory firmware is fixed to 1280 x 720 output.
What resolution did you set in the program you wrote?

Onzeisnottwice, 2023-05-19:
Thank you, I tried different resolutions. And also different settings for 
#define M5MODULEDISPLAY_PIXELCLOCK 356000000
//#define M5MODULEDISPLAY_REFRESH_RATE 66.0

The factory firmware works fine on the 5 inch version. Unfortunately, apparently I do not have that specific version. Ha Ha

None of them seems to work for the 5 inch version. I would be great if you still had exactly the same M5GFX lib that you used at the trade show. 
An additional note : when I say that nothin shows on the display, than I really mean nothing. Only the backlight is on. So somehow, I have the feeling that the 5 inch Waveshare is not really made alive (as in contrast to the 7 inch version).

lovyan03, 2023-05-19:
@Onzeisnottwice 

The log you presented showed a timing parameter.

[image]

You may be able to set this parameter manually so that it can be displayed.
Please try this code.
```
#include <M5AtomDisplay.h>

M5AtomDisplay display(1280, 720);

void setup(void)
{
  display.init();
  auto panel = (lgfx::Panel_M5HDMI*)display.getPanel();
  lgfx::Panel_M5HDMI::video_timing_t param;
  param.h.active = 1280;
  param.h.front_porch = 110;
  param.h.sync = 40;
  param.h.back_porch = 220;
  param.v.active = 720;
  param.v.front_porch = 5;
  param.v.sync = 5;
  param.v.back_porch = 20;
  panel->setVideoTiming(&param);
}

void loop(void)
{
  display.fillRect(rand() & 511, rand() & 511, 16, 16, rand());
  delay(1);
}
```

lovyan03, 2023-05-19:
Importantly, you do not need to change the pixel clock or refresh rate.
The factory demo is running at this resolution.
1280 x 720 @ 60Hz , pixel clock:742500000 .

Onzeisnottwice, 2023-05-19:
Sorry to say, but nothing shows up on the 5 inch with exactly your code. 7 inch runs smoothly. And again, the weird thing is that the display shows really nothing, not even scrambled graphics. I hope you have some other settings in your pocket. Ha Ha. (btw, I can't stress enough that the 5 inch works on the factory shipped AtomDislay). Also, I use the same cables throughout all my experiments, so a cable issue will most probably not the root cause.

lovyan03, 2023-05-19:
I understand. I will buy the same display and try it.

Onzeisnottwice, 2023-05-19:
Thank you so much. Please let us be updated on any progress you make.

lovyan03, 2023-05-24:
The WaveShare 5" HDMI panel I purchased had no problem displaying the screen.
It seems that the HDMI connector is in a different position than the one you have.
I seem to have bought a different product than what you have.....

I will order again....

[image]
[image]

Onzeisnottwice, 2023-05-24:
Thank you for trying this out. And very good news that you got something working. I also notice a VGA connector on your board. And on the side I also notice a number of buttons. Maybe yours is the NO TOUCHSCREEN version whereas the one I use includes touchscreen (which I am not going to use). But maybe I can order yours somewhere in Europe. Would you like to share an article number with us?  Thanks.

lovyan03, 2023-05-30:
@Onzeisnottwice 
Thank you for waiting,  I obtained that panel.

The cause is probably that the panel does not accept signals containing audio streams.
If you disable the audio stream and turn the power back on, it should work.

https://github.com/m5stack/M5GFX/blob/5495e8f8f31addfb23dc6bcb9effef4e95ff9173/src/lgfx/v1/panel/Panel_M5HDMI.cpp#L369-L379

From this section, comment out the "48kHz audio setting" line,
Then activate the "disable audio setting" line.
Importantly, audio is not disabled until the power is turned back on.

Are you wondering why I am enabling audio streaming on ATOMDisplay, which has no audio capability?
This is because this part is also common with ModuleDisplay and therefore requires audio functionality.
Also, I had no idea that there was such a ridiculous product that would not project if audio stream was enabled.

I am at a loss as to what to do to address this panel....

Onzeisnottwice, 2023-05-31:
Thank you so much. I can confirm that it works. 
Issue is solved.

closed 2023-11-20 by lovyan03

Onzeisnottwice, 2023-11-21:
Hi there, I am sorry to bother you but I need to open this ticket again. I tried the latest release of M5GFX on exact the same display as discussed in this thread. I disabled audio settings as you explained. But suddenly with this new release the 5 inch Waveshare display does not show anything anymore. I remember that you used the same type of display and came up with a solution. Could you please be so kind to check again? Many thanks.