m5stack/M5GFX #57: Initialization fails on models without LCDs.

ress997, 2023-07-17:
If I connect [Unit LCD](https://docs.m5stack.com/en/unit/lcd#example) or [Unit OLED]() to [AtomS3 Lite](https://docs.m5stack.com/en/core/AtomS3%20Lite) and run the posted example, it fails.

---

すみません日本語話者なので日本語で書かせていただきます。
AtomS3 LiteにUnit LCDもしくはUnit OLEDを繋いでサンプルコードを実行しようとしたらエラーが発生し実行することができません。
`#include <M5AtomS3.h>`を使用中は何も表示されなく、`M5UnitLCD display(2, 1, 400000);`に変更しても反応がありません。

## Error log (LCD)
```
In file included from /Users/ress/Arduino/libraries/FastLED/src/FastLED.h:75,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:8,
                 from /private/var/folders/3j/ytp_15tj0c5bw0tlz8kwm3dh0000gn/T/.arduinoIDE-unsaved2023618-92317-5augn8.v54ms/sketch_jul18b/sketch_jul18b.ino:1:
/Users/ress/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/device.hpp:42,
                 from /Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:18,
                 from /Users/ress/Arduino/libraries/M5GFX/src/M5UnitLCD.h:11,
                 from /private/var/folders/3j/ytp_15tj0c5bw0tlz8kwm3dh0000gn/T/.arduinoIDE-unsaved2023618-92317-5augn8.v54ms/sketch_jul18b/sketch_jul18b.ino:2:
/Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp: In member function 'void lgfx::v1::Bus_SPI::set_read_len(uint32_t)':
/Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:140:87: error: 'SPI_MISO_DLEN_REG' was not declared in this scope
     __attribute__ ((always_inline)) inline void set_read_len( uint32_t bitlen) { *reg(SPI_MISO_DLEN_REG(_spi_port)) = bitlen - 1; }
                                                                                       ^~~~~~~~~~~~~~~~~
/Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:140:87: note: suggested alternative: 'SPI_MS_DLEN_REG'
     __attribute__ ((always_inline)) inline void set_read_len( uint32_t bitlen) { *reg(SPI_MISO_DLEN_REG(_spi_port)) = bitlen - 1; }
                                                                                       ^~~~~~~~~~~~~~~~~
                                                                                       SPI_MS_DLEN_REG
In file included from /Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Setup.h:8,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI.h:28,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/M5Display.h:7,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:9,
                 from /private/var/folders/3j/ytp_15tj0c5bw0tlz8kwm3dh0000gn/T/.arduinoIDE-unsaved2023618-92317-5augn8.v54ms/sketch_jul18b/sketch_jul18b.ino:1:
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h: At global scope:
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:17:21: error: expected unqualified-id before numeric constant
 #define TFT_DISPOFF 0x28
                     ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:132:26: note: in expansion of macro 'TFT_DISPOFF'
     static constexpr int TFT_DISPOFF = 0x28;
                          ^~~~~~~~~~~
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:18:21: error: expected unqualified-id before numeric constant
 #define TFT_DISPON  0x29
                     ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:133:26: note: in expansion of macro 'TFT_DISPON'
     static constexpr int TFT_DISPON  = 0x29;
                          ^~~~~~~~~~
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:11:20: error: expected unqualified-id before numeric constant
 #define TFT_SLPIN  0x10
                    ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:134:26: note: in expansion of macro 'TFT_SLPIN'
     static constexpr int TFT_SLPIN   = 0x10;
                          ^~~~~~~~~
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:12:20: error: expected unqualified-id before numeric constant
 #define TFT_SLPOUT 0x11
                    ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:135:26: note: in expansion of macro 'TFT_SLPOUT'
     static constexpr int TFT_SLPOUT  = 0x11;
                          ^~~~~~~~~~

exit status 1

Compilation error: exit status 1
```
## Error log (OLED)
```
In file included from /Users/ress/Arduino/libraries/FastLED/src/FastLED.h:75,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:8,
                 from /private/var/folders/3j/ytp_15tj0c5bw0tlz8kwm3dh0000gn/T/.arduinoIDE-unsaved2023618-92317-1ivehie.2ccw/sketch_jul18a/sketch_jul18a.ino:1:
/Users/ress/Arduino/libraries/FastLED/src/fastspi.h:157:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/device.hpp:42,
                 from /Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:18,
                 from /Users/ress/Arduino/libraries/M5GFX/src/M5UnitOLED.h:11,
                 from /private/var/folders/3j/ytp_15tj0c5bw0tlz8kwm3dh0000gn/T/.arduinoIDE-unsaved2023618-92317-1ivehie.2ccw/sketch_jul18a/sketch_jul18a.ino:2:
/Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp: In member function 'void lgfx::v1::Bus_SPI::set_read_len(uint32_t)':
/Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:140:87: error: 'SPI_MISO_DLEN_REG' was not declared in this scope
     __attribute__ ((always_inline)) inline void set_read_len( uint32_t bitlen) { *reg(SPI_MISO_DLEN_REG(_spi_port)) = bitlen - 1; }
                                                                                       ^~~~~~~~~~~~~~~~~
/Users/ress/Arduino/libraries/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:140:87: note: suggested alternative: 'SPI_MS_DLEN_REG'
     __attribute__ ((always_inline)) inline void set_read_len( uint32_t bitlen) { *reg(SPI_MISO_DLEN_REG(_spi_port)) = bitlen - 1; }
                                                                                       ^~~~~~~~~~~~~~~~~
                                                                                       SPI_MS_DLEN_REG
In file included from /Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Setup.h:8,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI.h:28,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/M5Display.h:7,
                 from /Users/ress/Arduino/libraries/M5AtomS3/src/M5AtomS3.h:9,
                 from /private/var/folders/3j/ytp_15tj0c5bw0tlz8kwm3dh0000gn/T/.arduinoIDE-unsaved2023618-92317-1ivehie.2ccw/sketch_jul18a/sketch_jul18a.ino:1:
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h: At global scope:
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:17:21: error: expected unqualified-id before numeric constant
 #define TFT_DISPOFF 0x28
                     ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:132:26: note: in expansion of macro 'TFT_DISPOFF'
     static constexpr int TFT_DISPOFF = 0x28;
                          ^~~~~~~~~~~
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:18:21: error: expected unqualified-id before numeric constant
 #define TFT_DISPON  0x29
                     ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:133:26: note: in expansion of macro 'TFT_DISPON'
     static constexpr int TFT_DISPON  = 0x29;
                          ^~~~~~~~~~
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:11:20: error: expected unqualified-id before numeric constant
 #define TFT_SLPIN  0x10
                    ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:134:26: note: in expansion of macro 'TFT_SLPIN'
     static constexpr int TFT_SLPIN   = 0x10;
                          ^~~~~~~~~
/Users/ress/Arduino/libraries/M5AtomS3/src/utility/In_eSPI_Drivers/ST7789_Defines.h:12:20: error: expected unqualified-id before numeric constant
 #define TFT_SLPOUT 0x11
                    ^~~~
/Users/ress/Arduino/libraries/M5GFX/src/M5GFX.h:135:26: note: in expansion of macro 'TFT_SLPOUT'
     static constexpr int TFT_SLPOUT  = 0x11;
                          ^~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

ress997, 2023-07-17:
試しに手持ちの「Atom Lite」と「AtomS3 Lite」と「Unit LCD」と「Unit OLED」を組み合わせた結果を表に淡らしておきます

機材 | Unit LCD | Unit OLED
:---:|:---:|:---:
Atom Lite | 　スクロールしない | :ok:
AtomS3 Lite | 条件付き:ok: | 条件付き:ok:

- **スクロールしない** は「処」に部分を表示してスクロールしない
- **条件つき:ok:** は `display(2, 1, 400000)`の指定が必要だから

ress997, 2023-07-18:
まとめると問題点は次の3つです。
1. 「AtomS3 Lite」において`M5AtomS3.h`を使用中に`M5UnitOLED.h`もしくは `M5UnitOLED.h` が使用できない
2. `M5Unified.h`使用中に「Atom Lite」と「Unit LCD」を使用するとスクロールされない
3. `M5Unified.h`使用時は「AtomS3 Lite」はディスプレイが存在しないのに`display(2, 1, 400000)`の指定が必要で絵ある

lovyan03, 2023-07-18:
ご連絡ありがとうございます。
これから確認を進めてまいります。

取り急ぎ、M5Unifiedから使用する場合の例をお知らせいたします。

```cpp
#include <M5UnitLCD.h>
#include <M5UnitOLED.h>

// M5Unifiedより前にUnitLCD,UnitOLEDをincludeすると、M5.Displayとして利用できる。
// この場合は自分でインスタンスを作る必要はありません。

#include <M5Unified.h>

void setup(void) 
{
  M5.begin();

  M5.Display.println("Hello World !");

// M5.Displaysにインデクス番号を指定することでそれぞれ個別に操作できる。
  int idx_lcd = M5.getDisplayIndex(m5gfx::board_M5UnitLCD);
  if (idx_lcd >= 0) {
    M5.Displays(idx_lcd).print("UnitLCD");
  }

  int idx_oled = M5.getDisplayIndex(m5gfx::board_M5UnitOLED);
  if (idx_oled >= 0) {
    M5.Displays(idx_oled).print("UnitOLED");
  }

  delay(1000);
}

void loop(void)
{
  delay(1);
  M5.Display.fillCircle(rand() % M5.Display.width(), rand() % M5.Display.height(), 16, rand());
}
```

lovyan03, 2023-07-18:
M5GFXのdevelopブランチを更新しました。
次回のリリースに修正が反映される予定です。

今すぐ試したい場合はdevelopブランチから取得をお願いします。
このバージョンを使用するとビルドエラーは出なくなると思います。

ただし、AtomS3のボードマネージャ定義値に不備があるようでして、
SDAとSCLの値がPortAのピンとは異なるピンに設定されていることがわかりました。
この点につきましてはボードマネージャ管理担当の方に連絡しておきました。

いずれ修正されると思いますが、現時点ではボードマネージャの設定値は利用できないため、
ピン番号の設定を自前で行う必要があります。

includeより前にdefineでピン番号を設定することで動作させることができます。
```
#include <M5AtomS3.h>

#define M5UNITLCD_SDA 2
#define M5UNITLCD_SCL 1

#include <M5UnitLCD.h>
```