m5stack/M5GFX #60: StoreProhibited at FlipBuffer

yh1224, 2023-08-05:
Sometimes, acquiring DMA-capable memory fails, resulting in the following exception.

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e5cc4  PS      : 0x00060830  A0      : 0x800e98b5  A1      : 0x3ffd8b80
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000320  A5      : 0x3ffd8cc8
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000001
A10     : 0x3ffdc418  A11     : 0x00000140  A12     : 0x3f815024  A13     : 0x01000000
A14     : 0x00ffffff  A15     : 0x00000000  SAR     : 0x00000007  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4008b684  LEND    : 0x4008b69a  LCOUNT  : 0x00000000

Backtrace: 0x400e5cc1:0x3ffd8b80 0x400e98b2:0x3ffd8ba0 0x401db239:0x3ffd8c00 0x400e8ea2:0x3ffd8c20 0x400e48c5:0x3ffd8c60 0x400fec1a:0x3ffd8ca0 0x400ff592:0x3ffd8d30 0x400fe30b:0x3ffd8de0 0x400fe33d:0x3ffd8e40

  #0  0x400e5cc1:0x3ffd8b80 in unsigned int lgfx::v1::color_convert<lgfx::v1::swap565_t, lgfx::v1::bgr888_t>(unsigned int) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/misc/colortype.hpp:566
      (inlined by) unsigned int lgfx::v1::pixelcopy_t::copy_palette_affine<lgfx::v1::swap565_t, lgfx::v1::bgr888_t>(void*, unsigned int, unsigned int, lgfx::v1::pixelcopy_t*) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/misc/pixelcopy.hpp:188
  #1  0x400e98b2:0x3ffd8ba0 in lgfx::v1::Bus_SPI::writePixels(lgfx::v1::pixelcopy_t*, unsigned int) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:427
  #2  0x401db239:0x3ffd8c00 in lgfx::v1::Panel_LCD::writePixels(lgfx::v1::pixelcopy_t*, unsigned int, bool) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/panel/Panel_LCD.cpp:278
  #3  0x400e8ea2:0x3ffd8c20 in lgfx::v1::Panel_LCD::writeImage(unsigned int, unsigned int, unsigned int, unsigned int, lgfx::v1::pixelcopy_t*, bool) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/panel/Panel_LCD.cpp:346 (discriminator 4)
  #4  0x400e48c5:0x3ffd8c60 in lgfx::v1::LGFXBase::pushImage(int, int, int, int, lgfx::v1::pixelcopy_t*, bool) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/LGFXBase.cpp:1165
  #5  0x400fec1a:0x3ffd8ca0 in lgfx::v1::LGFX_Sprite::push_sprite(lgfx::v1::LovyanGFX*, int, int, unsigned int) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/LGFX_Sprite.hpp:540
  #6  0x400ff592:0x3ffd8d30 in lgfx::v1::LGFX_Sprite::pushSprite(lgfx::v1::LovyanGFX*, int, int) at .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/LGFX_Sprite.hpp:359
      (inlined by) m5avatar::Face::draw(m5avatar::DrawContext*) at .pio/libdeps/m5stack-core2/M5Stack-Avatar/src/Face.cpp:132
  #7  0x400fe30b:0x3ffd8de0 in m5avatar::Avatar::draw() at .pio/libdeps/m5stack-core2/M5Stack-Avatar/src/Avatar.cpp:155 (discriminator 4)
  #8  0x400fe33d:0x3ffd8e40 in m5avatar::drawLoop(void*) at .pio/libdeps/m5stack-core2/M5Stack-Avatar/src/Avatar.cpp:23
```

This is caused by the failure of allocating DMA-capable memory on FlipBuffer::getBuffer()
https://github.com/m5stack/M5GFX/blob/7370acbb2379b54f21ffeba70d72da9dbd5ec801/src/lgfx/v1/platforms/common.hpp#L146

called from Bus_SPI::writePixels()
https://github.com/m5stack/M5GFX/blob/7370acbb2379b54f21ffeba70d72da9dbd5ec801/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp#L427

I understand that this is due to the limitation of DMA-capable memory size and cannot be completely prevented from occurring due to conflicting with other processes, but I am wondering if there is some kind of countermeasure.

Please take it into consideration.

### Plan 1) Reduce the maximum memory size for one copy.

This can reduce the probability of failure. (but not perfect)

### Plan 2) Retry with a smaller size when the allocation fails.

This can increase the probability of succeed. (but it still might fail to allocate.)

### Plan 3) Skip the drawing process when the allocation fails.

The drawing result may be incomplete, but it can continue other processes.

### Plan 4) Throw exception immediately when the allocation fails.

This doesn't solve anything, but it makes it easier to figure out the reason for the reboot.

lovyan03, 2023-08-06:
@yh1224 
Thank you for the report. I will consider fix.

lovyan03, 2023-08-09:
Fixed in develop branch.
Changed to give up DMA transfers when memory is insufficient and use CPU time resources for transfers.

yh1224, 2023-08-09:
@lovyan03 I have confirm that the problem no longer occurs using the develop branch.
Thank you!

closed 2023-11-22 by lovyan03