m5stack/M5GFX #63: M5Stack CoreS3 is a supported device on framework ESP-IDF ?

GeorgeNegret, 2023-08-30:
I try to build M5GFX on Ubuntu 22.04 , framework ESP-IDF v5.1  device target esp32s3 but the build crash .
M5Stack CoreS3 is a supported device on framework ESP-IDF ?

Thanks !

GeorgeNegret, 2023-08-30:
I have never worked with M5GFX and ESP-IDF in combination.
Now I need M5Unified and M5GFX on ESP_IDF v5 framework because I want to run micro_ros_espidf_component on M5Stack CoreS3 .

lovyan03, 2023-08-30:
If you are considering using M5Unified for the first time with ESP-IDF, the following repositories and issues may be helpful.

https://github.com/hp27596/m5core2idfdemo
https://github.com/m5stack/M5Unified/issues/42

In addition, please note the following
 - The components to be used must be described in CMakeLists.txt file.
 - M5Unified and M5GFX are C++ libraries and cannot be used from .c source.

GeorgeNegret, 2023-08-30:
Sorry !  
I built M5GFX without M5Unified and build it's OK !
M5Unified it's my problem.

lovyan03, 2023-08-30:
I am glad to hear that you have not had any problems with M5GFX.
If you have problems with M5Unified, please write another Issue in the M5Unified repository.
However, maybe M5Unified is still not buildable with ESP-IDF v5.1...?

GeorgeNegret, 2023-08-30:
Thanks  @lovyan03 !

closed 2023-08-30 by GeorgeNegret