m5stack/M5GFX #64: M5Stack CoreS3 IDF 5.1.1 not compiling

oedze, 2023-09-07:
When using the M5Stack CoreS3 With IDF 5.1.1 a compile error pops up.

Line 809 of `src/lgfx/v1/platforms/esp32/common.cpp` is as follows: 
```cpp
 #if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 2) && ESP_IDF_VERSION < ESP_IDF_VERSION_VAL(5, 1, 0))
      (&dev->comd[0])[index].val = cmd_val;
 #else
      (&dev->comd0)[index].val = cmd_val;
 #endif
```
Which is checking for idf version 5, 0, 2 and higher and lower than 5.1.0. This does not work on 
Changing it to 
```cpp
 #if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 2) && ESP_IDF_VERSION <= ESP_IDF_VERSION_VAL(5, 1, 1))
      (&dev->comd[0])[index].val = cmd_val;
 #else
      (&dev->comd0)[index].val = cmd_val;
 #endif
```
fixes the issue
The ```&dev->comd[0]``` is correct here

lovyan03, 2023-09-08:
@oedze 
Thanks for the report !
I have fixed the develop branch, please try it and see.

oedze, 2023-09-21:
> @oedze Thanks for the report ! I have fixed the develop branch, please try it and see.

It works, thanks!

commit 7fd74595 (2023-09-08): fix #64
closed 2023-10-12 by lovyan03