m5stack/M5GFX #69: M5stack cording -

yukimach, 2023-10-31:
Hi, 
I'm trying to run the analogmeter example and hitting an error for the coreink. 

divide by zero error, could this be that the size of the screen is not recognised? 

---
[env:m5stack-coreink]
platform = espressif32
board = m5stack-coreink
framework = arduino
lib_deps = 
	m5stack/M5Core-Ink@^0.0.7
	m5stack/M5GFX@^0.1.10
monitor_speed =115200
monitor_filters = esp32_exception_decoder

--- Error
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d246a  PS      : 0x00060830  A0      : 0x800dcb39  A1      : 0x3ffb27f0  
A2      : 0x3ffc2564  A3      : 0x3ffc2560  A4      : 0x3ffc2584  A5      : 0x00000000  
A6      : 0x3ffbdb68  A7      : 0x00000000  A8      : 0x800d2463  A9      : 0x3ffb27a0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc255c  
A14     : 0x3ffc2584  A15     : 0x00000015  SAR     : 0x00000006  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008486d  LEND    : 0x40084875  LCOUNT  : 0x00000027  

Backtrace: 0x400d2467:0x3ffb27f0 0x400dcb36:0x3ffb2820

  #0  0x400d2467:0x3ffb27f0 in lgfx::v1::LGFXBase::width() const at .pio/libdeps/m5stack-coreink/M5GFX/src/lgfx/v1/LGFXBase.hpp:284
      (inlined by) setup() at src/main.ino:160
  #1  0x400dcb36:0x3ffb2820 in loopTask(void*) at /Users/yukimachida/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

lovyan03, 2023-10-31:
Hello, @yukimach 
I have tried it here and it works fine with no problems.

I have a puzzling point in your ini.
Why do you depend on the Core-Ink library?
Have you made any changes to the source code?
Have you removed the M5GFX init?

yukimach, 2023-11-01:
I removed the core-ink library now to avoid confusion. the core-ink library was working.
Espressif  platform 5.3.0. (I initially tried with 6 but also same problem). I've reloaded M5GFX a few times also to make sure. 

M5GFX display; 
display.begin();
has not been removed from the main file. 

---
[env:m5stack-coreink]
platform = espressif32
board = m5stack-coreink
framework = arduino
lib_deps = 

	m5stack/M5GFX@^0.1.10
monitor_speed =115200
monitor_filters = esp32_exception_decoder
---
I copy paste the code for analogmeter example, no modification.

lovyan03, 2023-11-01:
Hmmm,,, To check the situation, please run the following program with debug logging enabled.

```
#include <Arduino.h>
#include <M5GFX.h>
M5GFX display;

void setup(void)
{
  delay(100);
  display.begin();
  printf("w:%d h:%d\n", display.width(), display.height());
}

void loop(void)
{
  display.fillCircle(rand()%display.width(), rand()%display.height(), 16, rand());
  delay(4000);
}
```

Check the output of the serial monitor to see if M5GFX detects CoreInk.

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
[��mum����2-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   144][I][M5GFX.cpp:522] init_impl(): [M5GFX] [Autodetect] load from NVS : board:6
[   156][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:70 = 00f00000
[   157][I][M5GFX.cpp:682] autodetect(): [M5GFX] [Autodetect] M5StackCoreInk
w:200 h:200
```

yukimach, 2023-11-01:
Thanks! Yes I think it's this, I was trying to figure out how the library recognizes the board and dimensions, it looks like it's getting a 0 for width and height. 

w:0 h:0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d1b16  PS      : 0x00060a30  A0      : 0x800db21c  A1      : 0x3ffb2800  
A2      : 0x3ffc25c0  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000004  
A6      : 0x3ffb8874  A7      : 0x80000001  A8      : 0x800d1b0e  A9      : 0x3ffb27e0  
A10     : 0x5851f42d  A11     : 0x3ffb88d4  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0xffffffff  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4008488d  LEND    : 0x40084895  LCOUNT  : 0x00000027  

Backtrace: 0x400d1b13:0x3ffb2800 0x400db219:0x3ffb2820

  #0  0x400d1b13:0x3ffb2800 in lgfx::v1::LGFXBase::width() const at .pio/libdeps/m5stack-coreink/M5GFX/src/lgfx/v1/LGFXBase.hpp:284
      (inlined by) loop() at src/main.ino:14
  #1  0x400db219:0x3ffb2820 in loopTask(void*) at /.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

lovyan03, 2023-11-01:
Please enable debug logging.

platformio.ini
```
build_flags = -DCORE_DEBUG_LEVEL=5
```

And please post all logs prior to the error.

yukimach, 2023-11-01:
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
[    17][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   150][I][M5GFX.cpp:522] init_impl(): [M5GFX] [Autodetect] load from NVS : board:0
[   166][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:70 = 00f00100
[   180][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = 00000000
[   195][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:70 = 00f00100
[   209][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = 00000000
[   224][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:70 = 00f00100
[   238][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = 00000000
[   252][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:70 = 00f00100
[   266][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = 00000000
[   281][D][M5GFX.cpp:484] _read_panel_id(): [M5GFX] [Autodetect] read cmd:70 = 00f00100
[   282][I][M5GFX.cpp:589] init_impl(): [M5GFX] [Autodetect] save to NVS : board:0
w:0 h:0
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

lovyan03, 2023-11-01:
I apologies. I have found the cause of the problem and will correct it. Please wait a moment.

Have you recently purchased a CoreInk? It appears that a newer revision of the driver IC is being used than the one I own. As a result, there is a discrepancy in the determination results and the detection is failing.

I would change the decision formula so that the new driver IC does not cause an error.

yukimach, 2023-11-01:
Amazing, thank you so much. Yes I just purchased it last month I think.

lovyan03, 2023-11-01:
I have now updated the 'develop' branch.
Please check the operation with the develop version instead of the release version.

platformio.ini
```
lib_deps = https://github.com/m5stack/M5GFX.git#develop
```

If there are no problems, it will be included in the next update.

yukimach, 2023-11-01:
I manually changed it on my end and I confirm the fix works on my board now. 
The revision has yet to be released for platformio.

yukimach, 2023-11-01:
I confirm the develop branch also runs on my board.

closed 2023-11-01 by yukimach