m5stack/M5GFX #72: Question: drawPngFile using SdFat

maze61, 2023-11-11:
Hello @lovyan03,

in my M5 application, I depend on **long filenames** (8.3 filenames are really outdated).

So how can I use e.g. `drawPngFile `using the `SdFat` library? 

```
#include <FS.h>  // including this, didn't help
#include <SdFat.h>
#include <M5Unified.h>

SdFat32 sd; // for FAT16/FAT32
..
M5.Display.drawPngFile(sd, "/m5stack.png", 0, 0); 
```
fails with compilation errors:
```
...\Documents\Arduino\libraries\M5GFX\src/lgfx/v1/LGFXBase.hpp:888:23: error: cannot declare variable 'file' to be of abstract type 'lgfx::v1::DataWrapperT<SdFat32>'
       DataWrapperT<T> file ( &fs ); \
                       ^~~~
...\Documents\Arduino\libraries\M5GFX\src/lgfx/v1/LGFXBase.hpp:896:5: note: in expansion of macro 'LGFX_FUNCTION_GENERATOR'
     LGFX_FUNCTION_GENERATOR(drawPng, draw_png)
     ^~~~~~~~~~~~~~~~~~~~~~~
```

Thank you,
Marcus

lovyan03, 2023-11-21:
@maze61 

Thanks for reporting this.
I just updated the develop branch.

Until now, `SdFs` was supported, but not `SdFat32`.
With the latest develop branch, `SdFat32` will be available.
If there are no problems, this will be reflected in the next release.

maze61, 2023-11-21:
Hello @lovyan03,

these are excellent news, thank you so much!

Looking forward for the next release.

Take care, Marcus

closed 2023-11-21 by maze61