m5stack/M5GFX #73: Conflict error on LongTextScroll.ino

rydeen2060, 2023-11-14:
I'll share my experience.
I encountered an error when trying to run the LongTextScroll.ino sample code for M5GFX on M5CORE2.
The conflict error occurs at the bottom of M5GFX.h, specifically at the line:

cpp
Copy code
using EncodeRange = lgfx::EncodeRange;
I'm writing the code using VScode+PIO.
Interestingly, I don't get this error when using M5GFX in my custom code.
Moreover, the LongTextScroll.ino works fine on M5Stack Basic.

(JP)
わたしの体験を共有します。
M5GFXのサンプルコード**LongTextScroll.ino**をM5CORE2で動かそうとしたときのエラーについて。
M5GFX.hの一番したのほうの、
using EncodeRange = lgfx::EncodeRange;
でconflict errorが出てしまいます。
VScode+PIOで書いています。
自作コードでM5GFXを使っているとこのエラーは出ません。
また、M5Stack BasicではLongTextScroll.inoは普通に動きます。

lovyan03, 2023-11-20:
ご報告ありがとうございます。

Basicでは普通に動く一方で、Core2ではビルド時にコンフリクトエラーが出る、ということでしょうか？
条件の相違点が良く分からないので確認したいのですが、再現方法がよくわからないのでもう少し情報を頂けると助かります。
また、エラーの内容をそのまま見せて頂くことは可能でしょうか？

rydeen2060, 2023-11-20:
返信ありがとうございます。
今週は現場作業が続くので、週後半までに送ります。
前嶋武(rydeen2060)

rydeen2060, 2023-11-22:
らびあん様、

**１）platformio.iniのデータです。**

[env:m5stack-core2]
platform = espressif32
board = m5stack-core2
framework = arduino
lib_deps = 
	m5stack/M5Core2@^0.1.8
	m5stack/M5GFX@^0.1.11

**２）VScodeのバージョン情報です。**

バージョン: 1.83.1 (system setup)
コミット: f1b07bd25dfad64b0167beb15359ae573aecd2cc
日付: 2023-10-10T23:48:05.904Z
Electron: 25.8.4
ElectronBuildId: 24154031
Chromium: 114.0.5735.289
Node.js: 18.15.0
V8: 11.4.183.29-electron.0
OS: Windows_NT x64 10.0.22621

**３）ビルドした際に出た、エラーログです。**

In file included from src/main.cpp:3:

_.pio/libdeps/m5stack-core2/M5GFX/src/M5GFX.h:263:38: error: conflicting declaration 'using EncodeRange = struct lgfx::v1::EncodeRange'
 using EncodeRange = lgfx::EncodeRange;_

                                      ^
In file included from .pio/libdeps/m5stack-core2/M5Core2/src/utility/In_eSPI.h:480,
                 from .pio/libdeps/m5stack-core2/M5Core2/src/M5Display.h:10,
                 from .pio/libdeps/m5stack-core2/M5Core2/src/M5Core2.h:12,
                 from src/main.cpp:2:
.pio/libdeps/m5stack-core2/M5Core2/src/Fonts/GFXFF/gfxfont.h:26:3: note: previous declaration as 'typedef struct EncodeRange EncodeRange'
 } EncodeRange;
   ^~~~~~~~~~~
*** [.pio\build\m5stack-core2\src\main.cpp.o] Error 1
================= [FAILED] Took 13.52 seconds =================

以上です。

lovyan03, 2023-11-23:
なるほど、ありがとうございます。
サンプルコードを動作させるのに M5Core2.h は必要ないというか、
サンプルコードには include <M5Core2.h> の記述はなかったと思いますが、それはどこから生えてきましたか？

rydeen2060, 2023-11-23:
返信ありがとうございます。  <M5Core2.h>を外したら動きました。ありがとうございます。
必要ないものを書き込んでいたのですね。どうもすみませんでした。

closed 2024-02-28 by lovyan03