m5stack/M5GFX #75: readPixelRGB が M5Stack CoreS3 で機能しない

GOB52, 2024-02-09:
readPixelRGB does not work in M5Stack CoreS3.

## 構成
- M5Stack CoreS3

## ビルド環境
- macOS Catallina 10.15.7
- PlatformIO Core, version 6.1.13
- espressif32 / arduino
- M5Unified 0.1.12

## 問題の内容
Core,Core2 では readPixelRGB で値が取れているが、CoreS3 では取れていない。

### 再現コード
```cpp
#include <M5Unified.h>
auto& lcd = M5.Display;
void setup()
{
    M5.begin();

    lcd.clear(TFT_WHITE);
    uint16_t clr16 = lcd.readPixel(160,120);
    auto clr24 = lcd.readPixelRGB(160,120);
    M5_LOGI("rgb888 = 0x%02x%02x%02x, rgb565 = 0x%04x",clr24.r, clr24.g, clr24.b, clr16);
    // CoreS3     rgb888 = 0x000000, rgb565 = 0xffff
    // Core/Core2 rgb888 = 0xfcfcfc, rgb565 = 0xffff
}
void loop()
{
}
```

lovyan03, 2024-02-12:
報告ありがとうございました！
いま現象の原因を特定し、developブランチを更新しました。
こちらでは頂いた検証用コードで正しく動作するようになったことを確認いたしました。

GOB52, 2024-03-09:
あれ、0.1.13 で修正されてますよね?
Hash 8cc1b55 
自動で閉じてないのはなんでだろ?

GOB52, 2024-03-09:
閉じときます〜

closed 2024-03-09 by GOB52