m5stack/M5GFX #85: Continuous integration with GitHub actions

KenVanHoeylandt, 2024-03-10:
I was trying to implement CI to ensure the component compiles for the various versions of ESP-IDF.

The problem is that the project is not set up in a way that allows for this:
ESP-IDF components should be in a subfolder (commonly `main/`, but it could also be `M5GFX/`). Without this, any attempt to compile the component with `idf.py build` leads to this error:

> Called idf_component_register from a non-component directory.

Reference build: https://github.com/KenVanHoeylandt/M5GFX/actions/runs/8222729252/job/22484559726

**.github/workflows/build.yaml** example:
```yaml
name: Build
on: [push]
jobs:
  build-esp-idf-4.4:
    runs-on: ubuntu-latest
    steps:
      - name: Checkout repo
        uses: actions/checkout@v2
        with:
          submodules: recursive
      - name: build
        uses: espressif/esp-idf-ci-action@main
        with:
          esp_idf_version: v4.4
          target: esp32
```

To allow for `idf.py build` to work independently in this project, these steps are required:
- [ ] Add a subfolder for this component (`main/` or `M5GFX/`), move relevant files into this folder and add a `CMakeLists.txt`
- [ ] Add `include($ENV{IDF_PATH}/tools/cmake/project.cmake)` to the root `CMakeLists.txt`

Considering that moving source files has a big impact on your currently unmerged branches, I figured I'd open this issue to discuss the proposed changes.

lovyan03, 2024-03-29:
Hello, @KenVanHoeylandt 
Sorry, it's my lack of understanding and I don't quite understand what you're asking.
Are you saying you want the main source code of the library to be in `main` instead of `src`?

KenVanHoeylandt, 2024-10-19:
Almost: I suggest to move the `src` folder into a `main` folder. 
It doesn't have to be named "main" though, because ESP-IDF allows for renaming the module. "main" is just the default naming.

Having `src` in a folder allows us to create an ESP-IDF "module" (~ library).
I could make the `CMakeList.txt` files that would go along with it.

Once you have this module, you can compile it standalone, and thus we can avoid [releases with compile errors](https://github.com/KenVanHoeylandt/M5Unified/commit/43422fa2a307a8a4368b4dfcb4bac7de2055aee6) as we can automate building.

Sorry for the slow response.

closed 2025-05-16 by lovyan03