m5stack/M5GFX #86: M5StickCPlus2 display detection not handling ST7789V2

JustinOng, 2024-03-17:
Similar to https://github.com/m5stack/M5GFX/issues/69, I have a unit of the M5StickCPlus2 that is not being detected correctly.

Good unit:
```
[    42][D][M5GFX.cpp:482] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = 00528585
```

Bad unit:
```
[    40][D][M5GFX.cpp:482] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = ffb38181
```

Patching https://github.com/m5stack/M5GFX/blob/5342c2b4528260beddc62d5656717e80b8375554/src/M5GFX.cpp#L742 to also check for `0x81` solves the issue, though it baffles me as to how this unit can leave the factory without the SDK handling it properly.

lovyan03, 2024-03-18:
Hello, @JustinOng 
Thank you for letting me know this information. I just updated develop branch now. 
and, I will release an update soon.

Sorry. The factory firmware did not use M5GFX's automatic detection of the board and instead used a program that was specific to specific hardware, so no one noticed the problem.

I am maintaining the M5GFX from Japan, and I have not been notified of any changes made on the hardware manufacturing line in China. Therefore, no one, including me, was aware of the situation until your report.

JustinOng, 2024-03-18:
Thanks for the quick response!