m5stack/M5GFX #89: Display Module 13.2 not working with Core2 V1.1

mjau-scot, 2024-03-27:
I have an application that works fine when using Core2 V1.0 with the Display Module 13.2, but when I use the new Core2 V1.1, the display is blank.

M5.getDisplayIndex(m5gfx::board_M5ModuleDisplay) returns 1 on a Core2 V1.0, but returns -1 on a Core2 V1.1, so it appears the Core isn't seeing the display module.

lovyan03, 2024-03-29:
Hello, @mjau-scot 
Thank you for letting me know.
Updated develop branch. We'll be releasing updates soon, but if you want to try them out right away, please use the develop branch.

mjau-scot, 2024-03-29:
I can confirm that it's now working fine on Core2 v1.1 with the develop branch. Thanks for fixing this so quickly.

YouthInventor, 2024-10-31:
It's is not working for us as well. Do you have already solution of the problem?

lovyan03, 2025-11-19:
This issue appears to be resolved in the current version, so we will close it. If you have other issues, please create a new issue.