m5stack/M5GFX #94: LittleFS error occurs suddenly

sastles-com, 2024-05-20:
on my M5Dial, 

Using LittleFS, 
Suddenly this error occurs.

`.pio/libdeps/esp32/M5GFX/src/lgfx/v1/LGFXBase.hpp:920:23: error: cannot declare variable 'file' to be of abstract type 'lgfx::v1::DataWrapperT<fs::LittleFSFS>'
`
before then, this program normally runs.

sastles-com, 2024-05-20:
```
In file included from .pio/libdeps/esp32/M5GFX/src/M5GFX.h:22,
from .pio/libdeps/esp32/M5Unified/src/M5Unified.hpp:19,
from .pio/libdeps/esp32/M5Unified/src/M5Unified.h:5,
from src/main.cpp:2:
.pio/libdeps/esp32/M5GFX/src/lgfx/v1/LGFXBase.hpp: In instantiation of 'bool lgfx::v1::LGFXBase::drawJpgFile(T&, const char*, int32_t, int32_t, int32_t, int32_t, int32_t, int32_t, float, float, lgfx::v1::datum::datum_t) [with T = fs::LittleFSFS; int32_t = int; lgfx::v1::datum::datum_t = lgfx::v1::textdatum::textdatum_t]':
include/image.h:37:74: required from here
.pio/libdeps/esp32/M5GFX/src/lgfx/v1/LGFXBase.hpp:920:23: error: cannot declare variable 'file' to be of abstract type 'lgfx::v1::DataWrapperT[fs::LittleFSFS](javascript:void(0))'
```

It has been found that drawJpgFile is the problem.
Executing the following line of code:
M5Canvas canvas.drawJpgFile(LittleFS, "/image.jpg", 0, 0);
results in the error mentioned above.

sastles-com, 2024-05-20:
```
In file included from .pio/libdeps/esp32/M5GFX/src/M5GFX.h:22,
from .pio/libdeps/esp32/M5Unified/src/M5Unified.hpp:19,
from .pio/libdeps/esp32/M5Unified/src/M5Unified.h:5,
from src/main.cpp:2:
.pio/libdeps/esp32/M5GFX/src/lgfx/v1/LGFXBase.hpp: In instantiation of 'bool lgfx::v1::LGFXBase::drawJpgFile(T&, const char*, int32_t, int32_t, int32_t, int32_t, int32_t, int32_t, float, float, lgfx::v1::datum::datum_t) [with T = fs::LittleFSFS; int32_t = int; lgfx::v1::datum::datum_t = lgfx::v1::textdatum::textdatum_t]':
include/image.h:47:83: required from here
.pio/libdeps/esp32/M5GFX/src/lgfx/v1/LGFXBase.hpp:920:23: error: cannot declare variable 'file' to be of abstract type 'lgfx::v1::DataWrapperT[fs::LittleFSFS](javascript:void(0))'
DataWrapperT<T> file ( &fs ); \
^~~~
.pio/libdeps/esp32/M5GFX/src/lgfx/v1/LGFXBase.hpp:927:5: note: in expansion of macro 'LGFX_FUNCTION_GENERATOR'
LGFX_FUNCTION_GENERATOR(drawJpg, draw_jpg)
```

lovyan03, 2024-05-22:
Hello, @sastles-com 
I don't have your source code so I can only guess, but have you changed the order of your includes?
If you use LittleFS with M5GFXorM5Unified, you need to include the LittleFS headers first.

```
// include before M5Unified.h
#include <LittleFS.h>

// include after LittleFS.h
#include <M5Unified.h>

void setup(void) {
  M5.begin();
  LittleFS.begin();
  M5.Display.drawJpgFile(LittleFS, "/filename.jpg", 0, 0);
}
```

closed 2025-05-16 by lovyan03