m5stack/M5GFX #97: error compilation after upgrade library

yo2ldk, 2024-07-02:


yo2ldk, 2024-07-02:
I have it some example who work it good with Atom Lite Display HDMI, and now after update of library, I have many errors and cannot compile..  :(
 where can I find the latest library files before this new one ?
 
 `c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_get_png_row_default':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2815: multiple definition of `tdefl_get_png_row_default'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2815: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_compress':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2653: multiple definition of `tdefl_compress'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2653: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_compress_buffer':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2709: multiple definition of `tdefl_compress_buffer'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2709: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_init':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2714: multiple definition of `tdefl_init'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2714: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_get_prev_return_status':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2733: multiple definition of `tdefl_get_prev_return_status'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2733: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_get_adler32':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2738: multiple definition of `tdefl_get_adler32'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2738: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_compress_mem_to_output':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2743: multiple definition of `tdefl_compress_mem_to_output'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2743: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_compress_mem_to_heap':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2774: multiple definition of `tdefl_compress_mem_to_heap'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2774: first defined here
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_compress_mem_to_mem':
C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2783: multiple definition of `tdefl_compress_mem_to_mem'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\yo2ldk\Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2783: first defined here
[... cut ...]
c:/users/yo2ldk/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\M5GFX-master\lgfx\utility\miniz.c.o: in function `tdefl_write_image_to_png_file_in_memory':
C:\Users\ \Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/miniz.c:2880: multiple definition of `tdefl_write_image_to_png_file_in_memory'; libraries\M5GFX-master\lgfx\utility\lgfx_miniz.c.o:C:\Users\ \Documents\Arduino\libraries\M5GFX-master\src\lgfx\utility/lgfx_miniz.c:2880: first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board M5Stack-ATOM.
`

lovyan03, 2024-07-15:
Hello, @yo2ldk 

What version of M5GFX are you using?
The latest version is 0.1.16

closed 2025-04-01 by lovyan03

shanlb, 2025-04-25:
Same problem，seems like some functions in miniz.c (when include "miniz.sh") are already difined in lgfx/utility/lgfx_miniz.c