m5stack/M5GFX #98: Question: How can I draw a masked shape?

botamochi6277, 2024-07-02:
I want to draw a masked shape in display. For example, circle A masked by circle B in the following picture.
[image]

Would you have any ideas to draw the A AND B?

I have a rough idea using bool computation like this. I am worry about amount of memories with this approach. And, is it able with M5GFX?
```cpp
uint32_t img[HEIGHT][WIDTH]; 
uint32_t masked_img[HEIGHT][WIDTH]; 
uint8_t mask[HEIGHT][WIDTH]; 

fillCircle(img,10,50,100); // draw circle A
fillCircle(mask,50,10,100); // draw circle B
bitwise_and(img,mask,mask_img); // masking

display(masked_img);
```

lovyan03, 2024-07-15:
Hello, @botamochi6277 

Currently M5GFX does not have the functionality to handle bitmasks, so it must be implemented entirely in user code.

Other suggestions include the following functionality:

```cpp
#include <M5GFX.h>

M5GFX gfx;
M5Canvas cv;

void setup() {
    gfx.begin();

    cv.setColorDepth(2);
    cv.createSprite(160, 160);

    cv.drawCircle(50, 90, 50, 2);
    cv.drawCircle(90, 50, 50, 2);

    cv.floodFill(70, 70, 3);

    cv.setPaletteColor(3, TFT_YELLOW);

    cv.pushSprite(&gfx, 0, 0);
}

void loop() {}
```

[image]

botamochi6277, 2024-07-27:
Thanks for your reply. Sorry for the delay in commenting.
Oh, I got to know 🪣 `M5Canvas::floodFill` method.
Your proposed succeeds in my M5Stack and looks good.
I will try to use this approach. Thx.

closed 2024-07-27 by botamochi6277