m5stack/M5GFX #106: ESP-IDF master compilation crashes on one line

adokitkat, 2024-11-05:
Hello, I just wanted to report a compilation crash due to this line on ESP-IDF master branch:

https://github.com/m5stack/M5GFX/blob/ff11e09ac0bb5ec29433e81abbdc66c26488ba44/src/lgfx/v1/platforms/esp32/common.cpp#L677

`module` field was removed from `i2c_signal_conn_t` struct (missing from IDF v5.4 or v5.5, not sure).

lovyan03, 2024-11-05:
Hello, @adokitkat 
please try develop branch.

adokitkat, 2024-11-05:
That is a correct fix for previous targets, thanks!

I also want to say you may encounter a problem in a future because the code will probably not work for newer targets (P4, C61, etc.) as `periph_module_reset` should not be called outside i2c driver and is deprecated and doesn't do anything on new targets as far as I understand. 

We should fix this in the IDF though (new public APIs instead of you having to rely on private ones etc.). If you encounter something like this please open an issue in ESP-IDF repo, thanks.

suda-morris, 2024-11-05:
Hi @lovyan03 

For some reason (mostly because of hardware design changes), start from esp32p4, functions like `periph_module_disable` and `periph_module_reset` are not available. See: https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/include/esp_private/periph_ctrl.h#L74-L80

If you do want a way to reset the I2C peripheral, there's a low level function: `static inline void i2c_ll_reset_register(int i2c_port)`.

ainyan03, 2026-08-17:
Thank you for your patience — this is fixed in release 0.2.27. Please update and confirm; I believe the issue can then be closed.

The peripheral module is now selected explicitly per port, and the reset path uses `i2c_ll_reset_register()` / `lp_i2c_ll_reset_register()` as suggested in this thread, so the code no longer depends on the removed `i2c_periph_signal[].module` field and stays valid for chips where the mapping differs.