m5stack/M5GFX #107: A very interesting issue about board `autodetect`

Yixi, 2024-11-05:
Yesterday, after my three-year-old son dropped my Cardputer, I restarted it and found that the screen display was abnormal, with some black areas and some garbled sections.

[image]

At first, I thought the screen was damaged from the drop, but when I flashed other firmware, like M5Launcher, the screen displayed normally. Only the code I compiled based on M5Cardputer.h shows the abnormal display.

At this point, I began to suspect it was a screen configuration issue. So, I used the code  `Serial.printf("width:%d, height:%d\n", M5Cardputer.Display.width(), M5Cardputer.Display.height());` to print out the information, and found that both the screen width and height were 240.

However, I couldn’t find a place in M5GFX to manually configure the display settings. After extensive debugging, I discovered that during M5GFX’s autodetect process, the board was being identified as `board_t::board_M5VAMeter`.

The reason was that after I dropped the Cardputer, the value obtained in `result` at this point was `1`.

https://github.com/m5stack/M5GFX/blob/ff11e09ac0bb5ec29433e81abbdc66c26488ba44/src/M5GFX.cpp#L1475-L1486

I modified the code locally and after recompiling, the screen displayed normally.

[image]

[image]

This thing is so amazing...

So, is there a way for us to manually specify the board model and skip the autodetect logic?

lovyan03, 2024-11-05:
Hello @Yixi .

I think the IrDA LED was probably damaged by the impact of the fall. Since the M5GFX determines the model under the assumption that the hardware is not broken, it seems that it was no longer able to properly identify the model.
Currently, there is no way to disable automatic identification. I apologize for not being able to offer a solution right now.
I would like to consider some kind of countermeasure in the future.

lovyan03, 2025-01-27:
Hello, @Yixi 
M5GFX Version 0.2.4 has been released.
In this version, we have improved the procedure for identifying VAMeter and Cardputer.
Instead of judging the presence or absence of an IrDA LED as before, we now detect and determine the presence of an I2C device specific to VAMeter, so your Cardputer should be able to display the screen correctly.
Please give it a try.