m5stack/M5GFX #114: SD Card fails to initialize when using M5Stack Core2 v1.0 and Lora 868 v1.0 and M5GFX library with shared SPI bus

ifrew, 2025-01-23:
I upgraded to using the M5Unified.h library from the previous M5Core2.h library and now using V3 of the esp32 Ardunio core framework. M5Unified uses 0.2.1 and M5GFX is currently 0.2.0

When the Lora board is attached the SD fails to initialize. When not attached it works fine. The Lora pins used are 

#define SS_PIN      33 //5 for m5stack and 33 for m5core2 using Lora 1.0, 15 Lora v1.1 
#define RST_PIN     26  
#define DIO0_PIN    36
#define DIO1_PIN    -1
#define SPI_MOSI    23  // SPI MOSI pin
#define SPI_MISO    38  // SPI MISO pin
#define SPI_SCK     18   // SPI SCK pin

On using the espi_TFT library with the previous M5Core2.h library all works fine. I have a 4.7Kohm pullup resistor on the Lora CS pin 33 and I have set the pinmode to OUTPUT.

I had similar issues with the espi_TFT library because the SPI bus was shared but adding the pullup and setting the pinmode to output on the CS pin fixed that issue. 

I am assuming that this has to do with the shared SPI bus as I have seen many posts with similar issues. However. I do not see any pin conflicts here but I remember that this library does an autodetect and I am wondering if during that phase the SPI bus gets corrupted? 
Interestingly, this ONLY happens when power is applied for the first bootup. When I see the card failure, after removing and reinserting the sd card again and doing a reset of the device while power is still applied it see's the SD card and all works fine!!! However, at boot up this fails. This device is used for no user intervention and is in a emote location so I need it to work if power is lost and is resumed at a later time.

I've been working on this issue for over a week now with no success. Any insights appreciated.

Cheers

Iain

ifrew, 2025-01-24:
Update: Upon further review I believe the issue is that the CS Pin of the Lora board uses pin 33 for the SPI select but this is also the pin used by the Core2 for the External I2c Port I found. I don't see anywhere in my code where I initialize the I2C external port. In  the M5unified library this is not initialized by default. Does theM5GFX library initialize the external I2C port for the Core2 product? I don't think so looking at the code but there is a lot of checks going on. Hopefully the owner will know the answer.

lovyan03, 2025-01-27:
Hello, @ifrew 
Sorry, it's too much work to guess the source code from your text.
Please update all your libraries to the latest develop branch and provide the minimum buildable source code that reproduces your problem.

ifrew, 2025-01-28:
Going to close this out @lovyan03. I tried to repro it but couldn't as I only had the corev1.1 module available now and this seems to work although I have noticed that my USB power supply seem to be noisy so that's probably been the issue all along as sometimes the device won't boot. Sorry about that.

closed 2025-01-28 by ifrew