m5stack/M5GFX #120: m5paper how to use freefonts?

rudiratlos, 2025-03-16:
Hi,
i want to port from my old m5paper (M5EPD) to the new m5paperS3 (M5unified/m5gfx). 
in m5epd I've modified source code in ...Arduino/libraries/M5EPD/src/M5EPD_Canvas.h and added
`    
#define UTF8_ENABLE  2
#define CP437_ENABLE 1
void setAttribute(uint8_t attr_id, uint8_t param) {
        TFT_eSPI::setAttribute(attr_id, param);
    } 
    uint8_t getAttribute(uint8_t attr_id) {
        return TFT_eSPI::getAttribute(attr_id);
    } `

so I could use canvas.setAttribute(UTF8_ENABLE, false)?
I have modified my progmem fonts, to support ÄÖÜ?... characters, which I can use with UTF8_ENABLE=false

How can I set this attribute in M5unified??

lovyan03, 2025-03-16:
Hello, @rudiratlos 
Sorry, but that functionality does not exist in M5GFX.
Basically the M5EPD library was made by a different maintainer and is not compatible with the M5GFX library.

rudiratlos, 2025-03-18:
it works:

following this guideline:
[https://sigmdel.ca/michel/program/misc/gfxfont_8bit_en.html](https://sigmdel.ca/michel/program/misc/gfxfont_8bit_en.html)

` 
#define GFXFF 1
#define LOAD_GFXFF

#include "Fonts/OpenSans_Bold18pt8b.h"

#define  GOSB18b8 &OpenSans_Bold18pt8b

canvas[idx].setFreeFont(GOSB18b8);
  canvas[idx].setAttribute(utf8_switch, false); 
  canvas[idx].println("! \"#$%&'()*+,-./@^_`");
  canvas[idx].println("0123456789:;<=>?");
  canvas[idx].println("ABCDEFGVWXYZ");
  canvas[idx].println("abcdefgwxyz{|}[\\]~");
  canvas[idx].println(utf8toMyCharMap("L'été à €ÄÖÜäöüßÿ"));`

##############

`const uint8_t OpenSans_Bold18pt8bBitmaps[] PROGMEM = {
  0x00, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x7D, 0xE7, 0x9E, 0x79, 0xE7,
....
  0x00, 0x7F, 0xC0, 0x07, 0xF8, 0x00, 0x7E, 0x00, 0x00 };

const GFXglyph OpenSans_Bold18pt8bGlyphs[] PROGMEM = {
  {     0,   1,   1,   9,    0,    0 },   // 0x20 ' ' U+0020
  {     1,   6,  25,  10,    2,  -24 },   // 0x21 '!' U+0021
...
  { 10174,  18,  35,  22,    3,  -26 },   // 0xde 'LATIN SMALL LETTER THORN' U+00FE
  { 10253,  20,  35,  20,    0,  -26 } }; // 0xdf 'LATIN SMALL LETTER Y WITH DIAERESIS' U+000FF

const GFXfont OpenSans_Bold18pt8b PROGMEM = {
  (uint8_t  *)OpenSans_Bold18pt8bBitmaps,
  (GFXglyph *)OpenSans_Bold18pt8bGlyphs,
  0x20, 0xDF, 48 };`

closed 2025-03-21 by rudiratlos