m5stack/M5GFX #123: SD card not working

rudiratlos, 2025-03-17:
Hello,
I know that SD library is not included in m5unified/m5gfx. I tried:

SD.begin(4, SPI, 25000000); on my m5paper and
SD.begin(47, SPI, 25000000); on my new m5paperS3, but nothing worked.

my include order:
#include <SD.h>
#include <epdiy.h>
#include <M5Unified.h>

any suggestion, how get this working again?

felmue, 2025-03-21:
Hello @rudiratlos 

you probably need to initialize SPI with the proper GPIOs for M5PaperS3. E.g. something like:
```
SPI.begin(<SCK_PIN>, <MISO_PIN>, <MOSI_PIN>)
```
M5PaperS3 pinmap see [here](https://docs.m5stack.com/en/core/paperS3?id=pinmap).

Thanks
Felix

rudiratlos, 2025-03-23:
Thank you felmue, your tip was the right way. Below my code snipped to get SD.begin() working to read sd card file.

` 
  switch (M5.getBoard()) {
    case 19: // c_BoardVersionPaperS3
      PIN_SD_CS = GPIO_NUM_47;
      SPI.begin(GPIO_NUM_39, GPIO_NUM_40, GPIO_NUM_38, PIN_SD_CS); // SCK, MISO, MOSI, SD
    break;
    case 7: // c_BoardVersionPaper
      PIN_SD_CS = GPIO_NUM_4;
    break;
  }

lovyan03, 2025-09-08:
I think this issue is resolved so I'll close it.
Thanks @felmue .

closed 2025-09-08 by lovyan03