m5stack/M5GFX #128: AtomDisplay has no width or height

AcuarioCat, 2025-03-25:
Hi,
I am trying to get my AtomDisplay HDMI working without success.
The factory demo worked fine.
I have tried the basic example but I get nothing.
The odd thing is the display says it has 0 width and height.

Here is my code:
`
#include <Arduino.h>

#include <M5AtomDisplay.h>
M5AtomDisplay display(1280, 720, 50); // width, height

void setup(void)
{
    Serial.begin(115200);

    // initialize
    display.init();
    Serial.printf("W:%d H:%d\n", display.width(), display.height());

`
The serial output shows W:0 H:0 so it seems the display is not initialised correctly.

The same occurs when I try the default (no parameters), both are 0.

Any idea why this is happening?

Thanks

AcuarioCat, 2025-03-26:
I have been digging and enabled the ESP debug logging. I see a load of errors so I guess this is why. Is my AtomDisplay broken or an old version that is no longer supported?

`Opening port
Port open
[    16][D][esp32-hal-cpu.c:264] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32
  Package           : PICO-V3-02
  Revision          : 3.00
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x000000b3
  Embedded Flash    : Yes
  Embedded PSRAM    : Yes
  2.4GHz WiFi       : Yes
  Classic BT        : Yes
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   382032 B ( 373.1 KB)
  Free Bytes        :   351188 B ( 343.0 KB)
  Allocated Bytes   :    23632 B (  23.1 KB)
  Minimum Free Bytes:   345420 B ( 337.3 KB)
  Largest Free Block:   114676 B ( 112.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  8388608 B (8 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : DIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1984.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00200000, size:  1984.0 KB, type:  APP, subtype: OTA_1
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Mar 26 2025 11:22:10
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.4-972-g632e0c2a9f-dirty
  Arduino Version   : 3.0.7
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : esp32
  Arduino FQBN      : esp32:esp32:esp32:UploadSpeed=512000,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=no_fs,DebugLevel=debug,PSRAM=disabled,LoopCore=1,EventsCore=0,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default,fsupload=littlefs
============ Before Setup End ============
[   526][D][m5atomdisplay.h:153] init_impl(): [LGFX] pkg:6
[   531][I][panel_m5hdmi.cpp:506] init(): [M5HDMI] i2c port:1 sda:25 scl:21
[   539][I][panel_m5hdmi.cpp:511] init(): [M5HDMI] Chip ID: 17 02 e2
[   546][I][panel_m5hdmi.cpp:517] init(): [M5HDMI] Resetting HDMI transmitter...
[   560][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:81
[   569][I][panel_m5hdmi.cpp:500] _read_fpga_id(): [M5HDMI] FPGA ID:48440004
[   576][I][panel_m5hdmi.cpp:500] _read_fpga_id(): [M5HDMI] FPGA ID:48440004
[   583][D][panel_m5hdmi.cpp:714] _init_resolution(): [M5HDMI] PLL feedback_div:1  input_div:1  output_div:8  OUTPUT_CLOCK:74250000
[   594][D][panel_m5hdmi.cpp:715] _init_resolution(): [M5HDMI] logical resolution: w:1280 h:720
[   603][D][panel_m5hdmi.cpp:716] _init_resolution(): [M5HDMI] scaling resolution: w:1280 h:720
[   611][D][panel_m5hdmi.cpp:717] _init_resolution(): [M5HDMI]  output resolution: w:1280 h:720
[   620][D][panel_m5hdmi.cpp:718] _init_resolution(): [M5HDMI] video timing(Hori) total:1980 active:1280 frontporch:344 sync:26 backporch:330
[   632][D][panel_m5hdmi.cpp:719] _init_resolution(): [M5HDMI] video timing(Vert) total:750 active:720 frontporch:14 sync:1 backporch:15
[   644][I][panel_m5hdmi.cpp:580] init(): [M5HDMI] Initialize HDMI transmitter...
[   659][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   675][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:82
[   691][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:2b val:00
[   708][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:2e val:00
[   723][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   742][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   761][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   780][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   799][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   818][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   836][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   855][I][panel_m5hdmi.cpp:332] writeRegister(): [M5HDMI] i2c write err  reg:ff val:80
[   867][E][panel_m5hdmi.cpp:439] init(): [M5HDMI] failed to initialize the HDMI transmitter.
[   875][W][panel_m5hdmi.cpp:583] init(): [M5HDMI] HDMI transmitter Initialize failed.
W:0 H:0
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   382032 B ( 373.1 KB)
  Free Bytes        :   348164 B ( 340.0 KB)
  Allocated Bytes   :    26176 B (  25.6 KB)
  Minimum Free Bytes:   342020 B ( 334.0 KB)
  Largest Free Block:   114676 B ( 112.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     3 : UART_RX[0]
============ After Setup End =============
`

lovyan03, 2025-04-01:
Hello, @AcuarioCat 
There have been changes in the latest framework, so it may not be working properly.
Please downgrade your Arduino version to v2.x.x or try the development branch version.