m5stack/M5GFX #130: Calling drawBmpFile causes compilation error with SdFat objects

Phantom2x5, 2025-04-23:
### Relevant Details:
- Device -> Core2 IoT v1.1
- Arduino Framework
- SdFat 2.3.0
- M5Unified 0.2.5
- M5GFX 0.2.6

### Issue:
While attempting to call the drawBmpFile() function with the templated type SdFat32 as the referenced file object, compilation fails. Specifically, the errors are:
> .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/LGFXBase.hpp:949:23: error: cannot declare variable 'file' to be of abstract type 'lgfx::v1::DataWrapperT<SdFat32>'
       DataWrapperT<T> file ( &fs ); \

> .pio/libdeps/m5stack-core2/M5GFX/src/lgfx/v1/LGFXBase.hpp:955:5: note: in expansion of macro 'LGFX_FUNCTION_GENERATOR'
     LGFX_FUNCTION_GENERATOR(drawBmp, draw_bmp)

### Related Issue(s):
#72 #79 

### Code:
```#include <SPI.h>
#include <SdFat.h>
#include <M5Unified.h>

// Testing objects
SdFat32 sd;

// SD Card Pins
#define SD_MISO 38
#define SD_MOSI 23
#define SD_SCK 18
#define SD_CS 4

// SD Card Definitions for SdFat Library
#define SD_FAT_TYPE 1
#define SPI_CLOCK SD_SCK_MHZ(20)

// Arduino setup function that runs once from boot/reset
void setup() {

    SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);

    // Attempt to init SD card via chip select, then return result
    sd.begin(SD_CS, SPI_CLOCK);

    // Testing drawing of BMP file from SD Card
    M5.Display.drawBmpFile(sd, "/test.bmp", 0, 0);
}

// Arduino main loop function that runs continuously
void loop() {

}

Phantom2x5, 2025-04-23:
@lovyan03

lovyan03, 2025-04-30:
hello, @Phantom2x5 
It seems that you are missing M5.begin in your code.
Please execute M5.begin(); first.

lovyan03, 2025-04-30:
Sorry, @Phantom2x5 
It seems that the define SdFat_h macro was missing due to an update to SdFat.
GFX was using this to check whether SdFat was available, so it was no longer possible to compile.
I have made a correction to the develop branch, so please try it.

Phantom2x5, 2025-04-30:
@lovyan03 
No worries! I tried the fix and it works with SdFat now! My device (Core 2 IoT v1.1) is able to display pictures from the SD card.

closed 2025-04-30 by Phantom2x5