m5stack/M5GFX #141: In PlatformlO_SDL, the created window will not rotate with it.

Tinyu-Zhao, 2025-07-08:
Although the displayed content will rotate, the window width and height must be manually modified

https://github.com/m5stack/M5GFX/blob/13f0fe6f823d71cb6c863cdf27cfca42b82521f5/src/M5UnitOLED.h#L85-L88

Tinyu-Zhao, 2025-07-08:
When rotation is not enabled, the virtual screen is displayed vertically by default.

[image]

When rotation is turned on, although the content rotates, the virtual screen is still displayed vertically.

[image]

My expectation is that when the screen is rotated, the content rotates with it and the virtual screen can also be changed to horizontal display
I did this by cancelling the screen rotation, which would cause the text to rotate, and modifying the content of https://github.com/m5stack/M5GFX/blob/13f0fe6f823d71cb6c863cdf27cfca42b82521f5/src/M5UnitOLED.h#L82-L86 to

```cpp
      cfg.memory_width = 64;
      cfg.panel_width  = 64;
      cfg.memory_height = 128;
      cfg.panel_height  = 128;
```

[image]

lovyan03, 2025-07-16:
please push PC KEYBOARD "L" or "R". you can spin image.

Tinyu-Zhao, 2025-07-18:
> please push PC KEYBOARD "L" or "R". you can spin image.

Thanks for your help, I created a PR to help more people